f4exb/sdrangel #1641: [Feature Request] Show tracked object and rotator positions on polar/az-el plot in satellite and star tracker feature plugins

lrowen, 2023-03-29:
The current star tracker and satellite tracker plugins are great, but the az-el and polar displays are less useful than they could be.  I would like to be able to see the following three things in addition to the current track:
1.) The object's current (Now) or Custom position (or even both) along the track;
2.) The rotator control's commanded position;
3.) The rotator control's actual rotator returned position.

This would provide a more interactive display especially in the polar plot.  An example of what such could look like would be the GPredict rotator control screen's polar plot.

lrowen, 2023-03-29:
Ah, ok, I never noticed the text label moving with the track.... Could a larger-than-the-track-width dot be added at the current position, perhaps?

srcejon, 2023-03-29:
> We're a US not-for-profit educational site, and SDRangel has been absolutely wonderful for our use!

Care to share a link? Might be able to add some other stuff for you.

lrowen, 2023-03-29:
Sure; www.pari.edu

srcejon, 2023-03-29:
> Sure; [www.pari.edu](http://www.pari.edu)

Thanks. Very interesting. The HPBW figure in your screenshot suggests you might be using the 26m dish. That would be very cool!

lrowen, 2023-03-30:
Yes, and it is really cool.

srcejon, 2023-03-30:
> 1.) The object's current (Now) or Custom position (or even both) along the track;
> 2.) The rotator control's commanded position; 
> 3.) The rotator control's actual rotator returned position.

How about this? Red shows the target, yellow the current position while off target and then green when on target (To match the button in the controller GUI).

[image]

srcejon, 2023-03-30:
You mentioned on the other issue that you were looking at pointing error and active scans.

Currently a 2D map can be generated around a target from the Star Tracker (using an offset sweep):

[image]

I had planned to use this to calculate the pointing error - but never quite got around to it.

I hadn't thought of doing this using satellites, but it would be fairly straight forward to add the ability to target satellites as well.

closed 2023-08-30 by srcejon