f4exb/sdrangel #1642: In satellite tracker, voice synthesis is active for Custom time even if configured to be off for Now.

lrowen, 2023-03-29:
I am a frequent user of the Custom time in both the star tracker and satellite tracker plugins.  I have the voice synthesis turned off for satellite tracker, but when I use Custom time I still get the voice synthesis, and the voice is played every time the time changes in the Custom field.

srcejon, 2023-03-29:
Looks like when the settings keys were added, not all of the settings set by the settings dialog were included. Should be quick to fix. (There's probably a few other settings that aren't being set properly as well).

lrowen, 2023-03-29:
I'll pull that PR and check it out as soon as the drift scan of BIIF-4 is finished and I can do the rebuild.  Thanks!

srcejon, 2023-03-29:
GPS satellite? What are you scanning it for? :)

srcejon, 2023-03-29:
Antenna pattern calculator? Sounds like a good feature for SDRangel :)

lrowen, 2023-03-29:
I like it!  Anyway, that works.  No more voice.  I'd like to talk more about what a pattern calc plugin might look like, but this issue report is probably not the right place....perhaps on the mailing list....

srcejon, 2023-03-29:
> .perhaps on the mailing list....

Mailing list, forum or open another issue here. All good.

PR #1643 (2023-03-30): Correct settings keys for Satellite Tracker settings dialog. 
Fixes #1642
commit db0d0386 (2023-03-30): Correct settings keys for Satellite Tracker settings dialog. Fixes #1642
closed 2023-03-30 by f4exb