f4exb/sdrangel #1647: Satellite Tracker not pulling in TLE from new Celestrak format

lrowen, 2023-03-30:
Subject says it all.  The Celestrak website changed the URL format.  The page at [https://celestrak.org/NORAD/elements/](url) now points to a more dynamic query type URL.  Using the GPS operational satellites as an example, the URL is:

`https://celestrak.org/NORAD/elements/gp.php?GROUP=gps-ops&FORMAT=tle`

srcejon, 2023-03-31:
Seems to work for me using the commit tag you e-mailed.

In the log I see:

2023-03-31 08:40:34.034 (D) HttpDownloadManager: Downloading from  QUrl("https://celestrak.org/NORAD/elements/gp.php?GROUP=gps-ops&FORMAT=tle")  to  "/home/jon/.local/share/f4exb/SDRangel/tle_gp.php"
2023-03-31 08:40:34.226 (D) HttpDownloadManager: Download from  "https://celestrak.org/NORAD/elements/gp.php?GROUP=gps-ops&FORMAT=tle"  to  "/home/jon/.local/share/f4exb/SDRangel/tle_gp.php"  finshed.

[image]

What does it say in your log?

lrowen, 2023-03-31:
Hmm.... Not at the computer to check the log, unfortunately, but I do
remember seeing a tle_gp.php file.  If you have two of them (say
https://celestrak.org/NORAD/elements/gp.php?GROUP=gps-ops&FORMAT=tle and
https://celestrak.org/NORAD/elements/gp.php?GROUP=active&FORMAT=tle ) does
the download code overwrite tle_gp.php or merge it?  I was wondering why on
the production ops PC I had the 'active' the first, then the gps-ops list
second; the only satellites in the list are GPS.  On my laptop I had the
gps-ops first then the active, and it seems Celestrak has them named
differently in the two downloads.....  So perhaps a bug but not the bug I
thought..... I'll check when I have time, today's a full schedule and it
won't likely be today.

srcejon, 2023-03-31:
> If you have two of them (say https://celestrak.org/NORAD/elements/gp.php?GROUP=gps-ops&FORMAT=tle and https://celestrak.org/NORAD/elements/gp.php?GROUP=active&FORMAT=tle ) does the download code overwrite tle_gp.php or merge it? 

It's overwriting it, so I'll have to fix that.

PR #1673 (2023-04-28): Sat Tracker: Add query to TLE download filename, to avoid clashes. 
Fixes #1647
commit 02f264f2 (2023-04-28): Add query to TLE download filename, to avoid clashes. Fixes #1647
closed 2023-04-28 by f4exb