f4exb/sdrangel #1648: Satellite tracker crash on MacOs

loewal, 2023-04-01:
Compiled v.7.12 on MacOs Ventura.

Started with an empty workspace and added Satellite tracker
Crash with only: segmentation fault.

Did I start wrong...?

srcejon, 2023-04-12:
Adding Mac back trace posted in mailing list:

lldb) thread backtrace
 
 
* thread #22, name = 'QThread', stop reason = EXC_BAD_ACCESS (code=1, address=0x33fa8d30)
  * frame #0: 0x0000000100faf5e9 libsdrbase.dylib`ObjectPipesRegistrations::getPipes(QObject const*, QString const&, QList<ObjectPipe*>&) + 89
    frame #1: 0x00000001838c3610 libfeaturesatellitetracker.dylib`SatelliteTrackerWorker::update() + 3584
    frame #2: 0x0000000102ff78ef QtCore`___lldb_unnamed_symbol9624 + 159
    frame #3: 0x0000000102fe62e1 QtCore`QObject::event(QEvent*) + 113
    frame #4: 0x000000010282d886 QtWidgets`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
    frame #5: 0x000000010282ec0d QtWidgets`QApplication::notify(QObject*, QEvent*) + 477
    frame #6: 0x0000000102fbcd67 QtCore`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 167
    frame #7: 0x000000010301e2b5 QtCore`QTimerInfoList::activateTimers() + 1029
    frame #8: 0x000000010301c964 QtCore`QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 932
    frame #9: 0x0000000102fb9336 QtCore`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 502
    frame #10: 0x0000000102df406c QtCore`QThread::exec() + 140
    frame #11: 0x0000000102df4f60 QtCore`___lldb_unnamed_symbol7195 + 320
    frame #12: 0x00007ff80328a1d3 libsystem_pthread.dylib`_pthread_start + 125
    frame #13: 0x00007ff803285bd3 libsystem_pthread.dylib`thread_start + 15

And similar on ubuntu:

[image]

Both crashes appear to be in getPipes, rather than the Satellite Tracker. I've been unable to reproduce this on Windows, Linux or Mac. Do other plugins such as the AIS Feature crash?

Does the problem occur with the release .dmg as well as your own build?

srcejon, 2023-04-12:
If just your own build, make sure you do a:

    make clean

First, to ensure everything is rebuilt.

loewal, 2023-04-12:
I always start with a clean start and then git pull etc.
So…make clean, is that useful then?

loewal, 2023-04-12:
I use this script while in the Downloads folder:
clear
git clone https://github.com/f4exb/sdrangel.git
cd sdrangel
mkdir build
cd build
cmake -Wno-dev -DRX_SAMPLE_24BIT=ON ..
make -j2
sudo make install
cd ~/Downloads
sudo rm -R sdrangel

All plugins etc. are installed with Homebrew.
The satellite feature is the only one that crashes.

loewal, 2023-04-12:
Only my own, on Linux as well as Mac.
Ais is ok.

srcejon, 2023-04-12:
> I always start with a clean start and then git pull etc. So…make clean, is that useful then?

If you start from an empty directory, then no need for make clean.

srcejon, 2023-04-12:
Is it possible plugins from different builds are being mixed together? 

Maybe do:

    ls -l /opt/install/sdrangel/lib/sdrangel/plugins (or whereever you installed to)

And check the dates on the plugins match

loewal, 2023-04-12:
They are all from April 11.

On 12 Apr 2023 at 19:04 +0200, f4exb/sdrangel ***@***.***>, wrote:
>
> ls -l /opt/install/sdrangel/lib/sdrangel/plugins

srcejon, 2023-04-13:
Hmm. Built using your above script, and it works OK. What version of gcc are you using?

gcc --version
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0

srcejon, 2023-04-13:
Can you also confirm exactly when it crashes and the exact steps you use.

- Press Add feature
- Press start in Satellite Tracker... then crashes or before?
- Do you change any settings first?

loewal, 2023-04-13:
Hi Jon,

Thanks for investigating this.
DMG 13 (and also 12) work fine.

This is the Ubuntu-version:

gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

This is the MacOs-version:

Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: x86_64-apple-darwin22.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

On both systems (in fact 2x2=4) it crashes when..

• I start with an empty workspace
• Add the satellite-tracker
• Start it….segmentation fault!

Also adding the satellite-tracker to an existing preset crashes the same way

Regards

srcejon, 2023-04-13:
>  This is the Ubuntu-version: gcc (Ubuntu 12.2.0-3ubuntu1) 

Is it possible you built sgp4 library using gcc 11? If so, try rebuilding that.

loewal, 2023-04-13:
Hi Jon,

I compiled on both systems. Even rebooted.., but no, no difference...

srcejon, 2023-04-13:
Hmm. Running out of ideas.

What version of Qt is reported in SDRangel status bar?

loewal, 2023-04-13:
MacOs:
DMG: 5.15.2
Selfcompiled: 5.15.8

Ubuntu: 5.15.6

srcejon, 2023-04-13:
Try the debug build on ubuntu, instead. As I can more easily reproduce that. Thanks

loewal, 2023-04-13:
What is happening Jon?:

[ 37%] Linking CXX shared library ../../../lib/pluginssrv/liblocalsinksrv.dylib
Undefined symbols for architecture x86_64:
 "LocalSinkSettings::m_maxFFTBands", referenced from:
 LocalSinkSettings::deserialize(QByteArray const&) in localsinksettings.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/pluginssrv/liblocalsinksrv.dylib] Error 1
make[1]: *** [pluginssrv/channelrx/localsink/CMakeFiles/localsinksrv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 37%] Building CXX object pluginssrv/channelrx/filesink/CMakeFiles/filesinksrv.dir/filesinkmessages.cpp.o
[ 37%] Building CXX object pluginssrv/channelrx/filesink/CMakeFiles/filesinksrv.dir/filesinkwebapiadapter.cpp.o
[ 37%] Building CXX object pluginssrv/channelrx/filesink/CMakeFiles/filesinksrv.dir/filesinkplugin.cpp.o
[ 37%] Linking CXX shared library ../../../lib/pluginssrv/libfilesinksrv.dylib
[ 37%] Built target filesinksrv
make: *** [all] Error 2

This is new, never got this error.

srcejon, 2023-04-13:
It's just an issue when optimisation is disabled. I'll take a look tomorrow, but for now, you can skip it with cmake -DENABLE_CHANNELRX_LOCALSINK=OFF

loewal, 2023-04-14:
It gets (for me) even stranger..

The compilation on my Mac mini (Ubuntu) went fine. I will continue testing with the debugger.

However, why do I see this error only on my iMac. Here I will add the string you mention below and let you know.

loewal, 2023-04-14:
This is the result Jon:

[ 43%] Linking CXX shared library ../../../lib/pluginssrv/libmodaissrv.dylib
Undefined symbols for architecture x86_64:
 "AISModSettings::AISMOD_SAMPLE_RATE", referenced from:
 AISModSource::applySettings(AISModSettings const&, bool) in aismodsource.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/pluginssrv/libmodaissrv.dylib] Error 1
make[1]: *** [pluginssrv/channeltx/modais/CMakeFiles/modaissrv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 43%] Building CXX object pluginssrv/channeltx/modam/CMakeFiles/modamsrv.dir/ammodsettings.cpp.o
[ 43%] Building CXX object pluginssrv/channeltx/modam/CMakeFiles/modamsrv.dir/ammodbaseband.cpp.o
[ 43%] Building CXX object pluginssrv/channeltx/modam/CMakeFiles/modamsrv.dir/ammodsource.cpp.o
[ 43%] Building CXX object pluginssrv/channeltx/modam/CMakeFiles/modamsrv.dir/ammodwebapiadapter.cpp.o
[ 43%] Linking CXX shared library ../../../lib/pluginssrv/libmodamsrv.dylib
[ 43%] Built target modamsrv
make: *** [all] Error 2

The scripts are exactly the same on both machines and were always used without any problem.
This happens since adding the debug option to cmake.

Or….?

loewal, 2023-04-14:
This was the debugger-report on Ubuntu:

023-04-14 07:43:11.560 (I) WebAPIServer::start: starting web API server at http://:8091
Process 44536 stopped and restarted: thread 1 received signal: SIGCHLD
2023-04-14 07:43:32.019 (W) FeatureWebAPIUtils::mapSetDateTime: no Map feature
Process 44536 stopped
* thread #43, name = 'QThread', stop reason = signal SIGSEGV: address access protected (fault address: 0x7ffff5da6678)
 frame #0: 0x00007ffff5acd240 libQt5Core.so.5`___lldb_unnamed_symbol9613 + 48
libQt5Core.so.5`___lldb_unnamed_symbol9613:
-> 0x7ffff5acd240 <+48>: lock
 0x7ffff5acd241 <+49>: cmpxchgq %rdx, (%rdi)
 0x7ffff5acd245 <+53>: jne 0x7ffff5acd276 ; <+102>
 0x7ffff5acd247 <+55>: movq %rbp, 0x8(%rbx)
(lldb) thread backtrace
* thread #43, name = 'QThread', stop reason = signal SIGSEGV: address access protected (fault address: 0x7ffff5da6678)
 * frame #0: 0x00007ffff5acd240 libQt5Core.so.5`___lldb_unnamed_symbol9613 + 48
 frame #1: 0x00007ffff70c8c57 libsdrbase.so`QMutexLocker::QMutexLocker(this=0x00007fff991fd768, m=0x00007ffff7b376e8) at qmutex.h:233:43
 frame #2: 0x00007ffff70c910d libsdrbase.so`QMutexLocker::QMutexLocker(this=0x00007fff991fd768, m=0x00007ffff7b376e8) at qmutex.h:238:52
 frame #3: 0x00007ffff71e7437 libsdrbase.so`ObjectPipesRegistrations::getPipes(this=0x00007ffff7b37688, producer=0x0000555556323fe0, type=0x00007fff991fd8f8, pipes=0x00007fff991fd898) at objectpipesregistrations.cpp:145:32
 frame #4: 0x00007ffff71e6178 libsdrbase.so`MessagePipes::getMessagePipes(this=0x00007ffff7b37668, producer=0x0000555556323fe0, type=0x00007fff991fd8f8, pipes=0x00007fff991fd898) at messagepipes.cpp:50:36
 frame #5: 0x00007fffaf983ab9 libfeaturesatellitetracker.so`SatelliteTrackerWorker::update() + 2761
 frame #6: 0x00007ffff5cf7776 libQt5Core.so.5`___lldb_unnamed_symbol11852 + 118
 frame #7: 0x00007ffff5ce7945 libQt5Core.so.5`QObject::event(QEvent*) + 165
 frame #8: 0x00007ffff696bf32 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 130
 frame #9: 0x00007ffff5cbae38 libQt5Core.so.5`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 296
 frame #10: 0x00007ffff5d13e71 libQt5Core.so.5`QTimerInfoList::activateTimers() + 929
 frame #11: 0x00007ffff5d1475c libQt5Core.so.5`___lldb_unnamed_symbol11955 + 44
 frame #12: 0x00007ffff4121569 libglib-2.0.so.0`g_main_context_dispatch + 665
 frame #13: 0x00007ffff41763c8 libglib-2.0.so.0`___lldb_unnamed_symbol2758 + 520
 frame #14: 0x00007ffff411ed20 libglib-2.0.so.0`g_main_context_iteration + 48
 frame #15: 0x00007ffff5d14aea libQt5Core.so.5`QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 106
 frame #16: 0x00007ffff5cb97cb libQt5Core.so.5`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 299
 frame #17: 0x00007ffff5acba1b libQt5Core.so.5`QThread::exec() + 107
 frame #18: 0x00007ffff5accc51 libQt5Core.so.5`___lldb_unnamed_symbol9612 + 321
 frame #19: 0x00007ffff5290402 libc.so.6`start_thread(arg=<unavailable>) at pthread_create.c:442:8
 frame #20: 0x00007ffff531f590 libc.so.6`__clone3 at clone3.S:81

srcejon, 2023-04-14:
> It gets (for me) even stranger.. The compilation on my Mac mini (Ubuntu) went fine. I will continue testing with the debugger. However, why do I see this error only on my iMac. 

Probably because it's a different compiler (clang vs gcc) and has compiled that bit of code slightly differently.

loewal, 2023-04-14:
But that doesn’t explain why things compiled fine two days ago...

loewal, 2023-04-14:
???

I just plain old used "cmake -Wno-dev -DRX_SAMPLE_24BIT=ON"

And I don’t get any compilation-errors anymore.

Is there any logical explanation?

I wanted to compile myself because a) I can learn things and b) the DMG has no support for the Adam Pluto.

To make things easier: shall I make a pull-request for that?

loewal, 2023-04-14:
> It's because of the -DCMAKE_BUILD_TYPE=Debug option. That tells the compiler to compile slightly differently, which results in the error.
> If you use -DCMAKE_BUILD_TYPE=Release you wont see the problem. Note that not specifying -DCMAKE_BUILD_TYPE defaults to Release.

Forgot to ask:

So there is an error in the source code?

srcejon, 2025-01-25:
@loewal is this still available problem?

mattsoftware, 2025-04-22:
I just built the current master using the github ci using the mac_ci branch. As soon as it starts tracking a satellite the mac app just closes. I do not see a dump or anything. The debug log ends with this ... 

```2025-04-23 08:43:05.620 (D) SatelliteTracker::handleMessage: MsgConfigureSatelliteTracker
2025-04-23 08:43:05.620 (D) SatelliteTracker::applySettings: " m_latitude: -28.0744 m_longitude: 153.358 m_heightAboveSeaLevel: 0 m_target: NOAA 18m_satellites: ISS NOAA 15 NOAA 18 NOAA 19 NOAA 20 NOAA 9 NOAA-2 NOAA-21 JPSS-2 SONATE SONATE-2 CUTE CSC-1 ION SCV-002 ION SCV-012 PEARL-1H m_tles: https://db.satnogs.org/api/tle/ https://www.amsat.org/tle/current/nasabare.txt https://celestrak.org/NORAD/elements/gp.php?GROUP=weather&FORMAT=tle https://celestrak.org/NORAD/elements/gp.php?GROUP=gps-ops&FORMAT=tle https://celestrak.org/NORAD/elements/gp.php?CATNR=36395&FORMAT=tle m_dateTime:  m_minAOSElevation: 5 m_minPassElevation: 15 m_rotatorMaxAzimuth: 450 m_rotatorMaxElevation: 180 m_azElUnits: 1 m_groundTrackPoints: 100 m_dateFormat: yyyy/MM/dd m_utc: 0 m_updatePeriod: 1 m_dopplerPeriod: 10 m_predictionPeriod: 5 m_passStartTime: 00:00:00 m_passFinishTime: 23:59:59 m_defaultFrequency: 1e+08 m_drawOnMap: 1 m_autoTarget: 1 m_aosSpeech: ${name} is visible for ${duration} minutes. Max elevation, ${elevation} degrees. m_losSpeech: ${name} is no longer visible. m_aosCommand:  m_losCommand:  m_chartsDarkTheme: 1 m_replayEnabled: 0 m_replayStartDateTime:  m_sendTimeToMap: 1 m_dateTimeSelect: 0 m_mapFeature:  m_fileInputDevice:  m_drawRotators: 2 m_azimuthOffset: 0 m_elevationOffset: 0 m_columnSort: 3 m_columnSortOrder: 0 m_title: Satellite Tracker m_rgbColor: 4292942179 m_useReverseAPI: 1 m_reverseAPIAddress: 127.0.0.1 m_reverseAPIPort: 8888 m_reverseAPIFeatureSetIndex: 0 m_reverseAPIFeatureIndex: 0 m_workspaceIndex: 0"  force:  true
2025-04-23 08:43:05.662 (D) SatelliteTracker::readSatData - read  2271  satellites
2025-04-23 08:43:05.667 (D) LoadConfigurationFSM::restoreGeometry
2025-04-23 08:43:05.680 (W) SatelliteTracker::networkManagerFinished:  error( 1 ):  QNetworkReply::ConnectionRefusedError :  "Connection refused"```

The connection refused does not seem to be the reason. If I press play on the satellite window, and there is a sat to track as I hit play, the app just crashes. This is my latest log .... 

```2025-04-23 08:50:52.660 (I) SDRangel 7.22.6.msw-7-g1c8d7bdaf Qt 6.7.3 64b arm64 macOS Sequoia (15.2) DSP Rx:24b Tx:16b PID 70330```

(all I did was open the app, and press play on the satellite feature windows)

Is this a separate issue? I'm not sure it matches the current thread, but may be related. For the recod it seems to work fine for me on the latest stable release downloaded from the site. This is the current master build running on my m2 mac.

mattsoftware, 2025-04-22:
My issue might be satellite related. As it seems to be if i just load a satellite thats appearing soon is enough to crash it. But some satellites do not seem to crash it. Here is the log for adding a single satellite, and as soon as I hit play it crashes...

```
2025-04-23 09:02:07.374 (D) SatelliteTrackerWorker: Current time:  "2025-04-22T23:02:07.373Z"
2025-04-23 09:02:07.374 (D) SatelliteTrackerWorker: New AOS:  "SITRO-AIS-6"  new:  QDateTime(2025-04-22 22:57:46.000 UTC Qt::UTC)  old:  QDateTime(Invalid)
2025-04-23 09:02:07.374 (D) SatelliteTrackerWorker: New LOS:  "SITRO-AIS-6"  new:  QDateTime(2025-04-22 23:06:55.000 UTC Qt::UTC)  old:  QDateTime(Invalid)
2025-04-23 09:02:07.374 (D) SatelliteTrackerWorker::aos  "SITRO-AIS-6"
2025-04-23 09:02:07.374 (D) SatelliteTrackerWorker::aos - autoTarget setting  "SITRO-AIS-6"
2025-04-23 09:02:07.387 (D) SatelliteTrackerWorker:: Interval to LOS  287627
2025-04-23 09:02:07.387 (D) SatelliteTrackerWorker: Current time:  "2025-04-22T23:02:07.373Z"
2025-04-23 09:02:07.387 (D) SatelliteTrackerWorker: New AOS:  "ISS"  new:  QDateTime(2025-04-23 03:55:05.000 UTC Qt::UTC)  old:  QDateTime(Invalid)
2025-04-23 09:02:07.387 (D) SatelliteTrackerWorker: New LOS:  "ISS"  new:  QDateTime(2025-04-23 04:03:43.000 UTC Qt::UTC)  old:  QDateTime(Invalid)
2025-04-23 09:02:07.387 (D) SatelliteTrackerWorker:: Interval to LOS  18095627
2025-04-23 09:02:07.411 (D) SatelliteTracker::handleMessage: MsgConfigureSatelliteTracker
2025-04-23 09:02:07.411 (D) SatelliteTracker::applySettings: " m_target: SITRO-AIS-6"  force:  false
2025-04-23 09:02:07.411 (D) SatelliteTrackerWorker::applySettings: " m_target: SITRO-AIS-6"  force:  false
2025-04-23 09:02:07.435 (W) SatelliteTracker::networkManagerFinished:  error( 1 ):  QNetworkReply::ConnectionRefusedError :  "Connection refused"
```

So I'm thinking the connectin refused might be something worth looking at. Just adding it here as an interesting note. Although I am feelig like this might be a separate issue.