f4exb/sdrangel #1653: Cariboulite SDR

KH6JAG, 2023-04-07:
Can the Cariboulite SDR be use with SDR angel?
https://github.com/cariboulabs/cariboulite

f4exb, 2023-04-08:
I see this can work with Soapy SDR so presumably the Soapy SDR plugins may be used. Note that they are not loaded by default (for compatibility issues with some device specific plugins) and you have to use the `--soapy` option on the command line to "activate" Soapy SDR support.

Of course  a device specific plugin would be preferable (I think they provide an API) but requires some development.

Also the best implementation of it I think would be to run headless on the Raspberry Pi and transfer the samples over the network to a laptop or desktop running SDRangel in GUI mode using a RemoteSink or RemoteTCP sink Rx channel on the Pi and the RemoteInput or RemoteTCPInput input plugins on the laptop or desktop. For Tx only the RemoteSource Tx plugin and RemoteOutput plugins exist that use UDP for transfer.

KH6JAG, 2023-04-08:
Thank you for the advice.  I will give it a try - I'm new to this unit but I have used the LimeSDR mini and I was hoping to potentially have similar capabilities as this is also a transceiver.  One idea was to see if it was possible to use as a Portsdown DATV unit,

f4exb, 2023-04-09:
It looks like an interesting project and I have pre-ordered one. I'll see what I can do about it...

j1m1l0k0, 2023-06-14:
does it work capturing KU band? satellite start one D2/C4 - 70.0W

loewal, 2023-08-05:
Hi all,
I bought the Cariboulite, installed it on a Raspberry Pi3B+, and also installed SDR++ and run this as server with "sdrpp -s". Runs fine on my Mac Mini with SDR++.
Then I tried SDRangel, but discovered that I need the 'Soapy-enabled version'.
So, managed to compile an executable, but this crashes with an fatal exeption. The nonsoapy-version behaves the same.
The last DMG however runs perfectly....and you'll guess: no Soapy support.
Is it possible that one of you compiles a DMG with Soapysupport?

Thank you1

f4exb, 2024-05-29:
Cariboulite has no native support. I entirely relies on SoapySDR which is already supported (yet not by default) in SDRangel. So this issue can be closed.

closed 2024-05-29 by f4exb