f4exb/sdrangel #1656: RTL-SDR missing from receiver list

cjcfojc, 2023-04-11:
I am running Ubuntu Mate 22.04, Kernel 5.19.0-38.  SDR Angel runs flawlessly on my system, however, none of my RTL-SDR V3 dongles show up in receiver list (they show up in the device list).  I did a test with two dongles, and i tested all of the builds from 2023 (jan through now) and get the same issue.  OCCASIONALLY, and it seems to be quite random, one, or the other will come up; occasionally my "second" dongle will, and occasionally either or both will be available running sudo sdrangel from terminal (again, this is just my process trying to figure out the issue).  For said test, i setup the ADS-B module, again, successfully, but that success is intermittent depending on whether or not the app recognizes the dongles.  I have also tried compiling SDR Angel from scratch, and have done so successfully, yet the issue remains.  any guidance or help would be appreciated!  just to add some more info, i loaded up SDR++, SDRTrunk, and a few other apps and both dongles are present when selecting receivers.

Just to clarify, this has been my process so far...

- tested all SDR Angel builds (the ubuntu 22.04 .deb packages ) from 2023
- compiled the latest, from scratch, with zero errors or dependency issues, to include all the hardware compiling in the instructions
- ran the app with superuser
- tested my hardware in several other apps

current version of SDR Angel installed is 7.13

Again, through my own testing, i have installed each of these builds in various ways via gdebi, package installer, terminal, and software center, but still my RTL-SDR dongles are not present as an RX device in the list

Ccd513, 2023-04-11:
Mine was working fine the other day and now I'm having the same issue

srcejon, 2023-04-12:
> Mine was working fine the other day and now I'm having the same issue

I don't think anything within SDRangel related to the RTL SDR has changed for quite a while. Has anything changed with your computer / OS?

srcejon, 2023-04-12:
> I am running Ubuntu Mate 22.04, Kernel 5.19.0-38. SDR Angel runs flawlessly on my system, however, none of my RTL-SDR V3 dongles show up in receiver list (they show up in the device list). 

What do you mean by 'device list'?

Can you attach the SDRangel log file to this issue please?

cjcfojc, 2023-04-12:
> > I am running Ubuntu Mate 22.04, Kernel 5.19.0-38. SDR Angel runs flawlessly on my system, however, none of my RTL-SDR V3 dongles show up in receiver list (they show up in the device list).
> 
> What do you mean by 'device list'?
> 
> Can you attach the SDRangel log file to this issue please?

I will post the log when I'm off work.  The device list as in choosing a receiver.  The only change that has happened to my OS would be LTS updates.  I will definitely post the log asap.

srcejon, 2023-04-12:
Sorry, I'm still not sure what you mean by receiver list and device list. By device list, do you mean the list of plugins under Help > Loaded plugins? And by receiver list, do you mean the list of sampling devices you get when you press the Add Rx Device button:

[image]

This is working for me on the same kernel: 5.19.0-38-generic #39~22.04.1-Ubuntu

cjcfojc, 2023-04-12:
Yes.  None of my RTLSDR devices show up there.  As I stated in the original post, I've used 5 different builds, and did a complete build from source with no errors, and they don't show up.  SDRPP is good to go, SDRTrunk, GQRX, everything else I use, my devices are recognized.  I will be sure to post the log when I get home.

srcejon, 2023-04-12:
One thing also to check is which librtlsdr.so you are using. So:

    ldd /opt/install/sdrangel/lib/sdrangel/plugins/libinputrtlsdr.so  | grep rtl

Does it point to the one you built for sdrangel, which should be:

    librtlsdr.so.0 => /opt/install/librtlsdr/lib/librtlsdr.so.0 

Or something else installed by other s/w?

cjcfojc, 2023-04-12:
i'm currently running the deb from the 2053_master and all dependencies are met.  i removed all previous installations of SDRAngel including the source compiled version.

cjcfojc, 2023-04-12:
When i launch SDR Angel from terminal, it definitely recognizes the RTL-SDR:

2023-04-12 18:14:43.435 (D) RTLSDRPlugin::enumSampleSources: enumerated RTL-SDR device #1357 

and

Found Rafael Micro R820T tuner
2023-04-12 18:14:47.490 (I) RTLSDRInput::openDevice: open: Realtek RTL2838UHIDIR, SN: 00000001

[image]

cjcfojc, 2023-04-12:
When i launch SDR Angel from terminal, it definitely recognizes the RTL-SDR:

2023-04-12 18:14:43.435 (D) RTLSDRPlugin::enumSampleSources: enumerated RTL-SDR device

and

Found Rafael Micro R820T tuner
2023-04-12 18:14:47.490 (I) RTLSDRInput::openDevice: open: Realtek RTL2838UHIDIR, SN: 00000001

[image]

f4exb, 2023-04-13:
As far as I understand you have several RTL-SDRs connected to your system in which case if you want all of them recognized correctly in SDRangel you have to give them different serial numbers. You have to use the `rtl_eeprom` utility that comes with librtlsdr. See this video around the 3rd minute: https://youtu.be/pgzyT0B8ZQ8

srcejon, 2023-04-13:
> Found Rafael Micro R820T tuner 2023-04-12 18:14:47.490 (I) RTLSDRInput::openDevice: open: Realtek RTL2838UHIDIR, SN: 00000001

You only get this message when the RTL SDR is opened. If the device is already opened, it will not appear in the device list.

Can you share a picture of your full screen please? How many workspaces do you have?

cjcfojc, 2023-04-13:
Nothing happens.  I have no work spaces because I can't select my sampling device as it is not listed.  Also, I have already renamed all of my dongles.

srcejon, 2023-04-13:
> Nothing happens. I have no work spaces because I can't select my sampling device as it is not listed. Also, I have already renamed all of my dongles.

You must have a workspace, otherwise you wouldn't be able to press the add Rx device button.

And you'll only get the message: RTLSDRInput::openDevice: open: when a device is opened.

Please do Workspaces > View all and share a picture of your screen. Maybe something is hidden.

cjcfojc, 2023-04-13:
[image]

srcejon, 2023-04-13:
The scroll bars suggest there is something hidden to the right and bottom - can you scroll to check?

cjcfojc, 2023-04-13:
Man do I feel dumb...I've got this giant wide screen and I didn't even think to scroll over there.  I really appreciate everyone's help, and definitely patience.  Attached are two photos my girlfriend just took because I've been screwing over this since I left for work this morning hahahaha.  Sorry for the crappy quality.  Again, thank you guys for your help.  I really appreciate it!

I will leave this issue open through the day until I can physically confirm that it's squared away.

[image]
[image]

cjcfojc, 2023-04-13:
This issue is resolved.  I appreciate the help i received and thanks for your patience .

closed 2023-04-13 by cjcfojc