f4exb/sdrangel #1657: Support Adalm Pluto

loewal, 2023-04-14:
Sorry if this is not the right place:

I would like support fot the Adalm Pluto in the MacOs DMG-version.

Thanks

srcejon, 2023-04-14:
Currently ENABLE_IIO=OFF for the DMG build, because current build of libxml2 requires autoconf which is a dependency I didn't want on Mac.

However, it looks like it may be able to be built with cmake instead - so I'll give that a try.

loewal, 2023-04-14:
Thanks in advance Jon

Regards

srcejon, 2023-04-14:
Give this a try - No idea if it works, as I don't have a pluto.

https://sdrangel.org/downloads/sdrangel-7.13.0_10.14.6_x86_64.dmg

loewal, 2023-04-14:
Yes Jon. It works!
Ever so grateful. Will support be permanent now?

Kind regards

srcejon, 2023-07-03:
Implemented.

closed 2023-07-03 by srcejon