f4exb/sdrangel #1660: Cannot load library liblocalsink

Tungsten842, 2023-04-15:
When starting sdrangel, built from source on nix:
```
(W) PluginManager::loadPluginsDir: Cannot load library /nix/store/l4mfknw71klgmmca1nj8ipzj135yas67-sdrangel-7.13.0/lib/sdrangel/plugins/liblocalsink.so: (/nix/store/l4mfknw71klgmmca1nj8ipzj135yas67-sdrangel-7.13.0/lib/sdrangel/plugins/liblocalsink.so: undefined symbol: _ZN17LocalSinkSettings13m_maxFFTBandsE)
```
These are the compilation logs:
[log.txt](https://github.com/f4exb/sdrangel/files/11239012/log.txt)

Artemia76, 2023-04-16:
Same here on Archlinux

Tungsten842, 2023-05-27:
Still a problem

srcejon, 2023-05-27:
I guess there's something odd with gcc 12.

In localsinksettings.h change:

    static const uint32_t m_maxFFTBands = 20;

To

    static const uint32_t m_maxFFTBands;

And in localsinsettings.cpp. add:

const uint32_t LocalSinkSettings::m_maxFFTBands = 20;

And see what happens.

Tungsten842, 2023-05-28:
> I guess there's something odd with gcc 12.
> 
> In localsinksettings.h change:
> 
> ```
> static const uint32_t m_maxFFTBands = 20;
> ```
> 
> To
> 
> ```
> static const uint32_t m_maxFFTBands;
> ```
> 
> And in localsinksettings.cpp. on line 25 add:
> 
> ```
> const uint32_t LocalSinkSettings::m_maxFFTBands = 20;
> ```
> 
> And see what happens.

It works!

Tungsten842, 2023-06-10:
@srcejon Is there any plan to incorporate this fix in to the next release?

srcejon, 2023-06-10:
Well, I'm not sure that really should be the fix - as the original code looks ok.

srcejon, 2023-06-30:
> I guess there's something odd with gcc 12.

Seems to work ok using gcc 12.2.0 on Ubuntu 23.04 x64.

f4exb, 2023-07-06:
I am reluctant to change code that is syntactically correct and follows C++ rules. It is a compiler bug and patches are already being distributed. Moreover this pattern can be found all over the place and could introduce problems you don't see or don't see yet. Please make sure you use the latest version of your compiler.

Tungsten842, 2023-07-06:
> I am reluctant to change code that is syntactically correct and follows C++ rules. It is a compiler bug and patches are already being distributed. Moreover this pattern can be found all over the place and could introduce problems you don't see or don't see yet. Please make sure you use the latest version of your compiler.
> 
> If you are concerned by the LocalSink plugin then patch the code on your clone or branch,

I have tested some compilers:
Same bug with gcc 12.3, 13.1
Sdrangel doesn't compile with gcc 10.4, 11.4.0 and clang 16.0.6
It works with clang 14.0.6, 15.0.7

srcejon, 2023-07-06:
>Sdrangel doesn't compile with gcc 10.4, 11.4.0 and clang 16.0.6 

What errors are you getting with these?   10.2 and 11.3.0 are fine for me.

Tungsten842, 2023-07-06:
> > Sdrangel doesn't compile with gcc 10.4, 11.4.0 and clang 16.0.6
> 
> What errors are you getting with these? 10.2 and 11.3.0 are fine for me.

Clang 16.0.6: [clang.txt](https://github.com/f4exb/sdrangel/files/11972115/clang.txt)
Gcc 11.4:  [gcc11.txt](https://github.com/f4exb/sdrangel/files/11972147/gcc11.txt)
Gcc 10.4: [gcc10.txt](https://github.com/f4exb/sdrangel/files/11972152/gcc10.txt)

srcejon, 2023-07-06:
The clang issue is because the gold linker plugin is missing - perhaps you need to install it?
The gcc failures also don't look to be due to SDRangel - but a gcc/binutils/libstdc++ problem.

closed 2023-08-22 by github-actions[bot]

RobRich999, 2023-11-10:
I can patch it locally or build with LLVM, so no biggie here, though I can report the undefined symbol issue is happening with the current GCC 13.2.0 release as well.

srcejon, 2023-11-10:
> I can patch it locally or build with LLVM, so no biggie here, though I can report the undefined symbol issue is happening with the current GCC 13.2.0 release as well.

What OS are you building on?

RobRich999, 2023-11-11:
Kubuntu 23.10 with GCC 13.2.0 (Ubuntu 13.2.0-4ubuntu3).

The warning during sdrangel startup:

`Cannot load library /opt/install/sdrangel/lib/sdrangel/plugins/liblocalsink.so: (/opt/install/sdrangel/lib/sdrangel/plugins/liblocalsink.so: undefined symbol: _ZN17LocalSinkSettings13m_maxFFTBandsE)`