f4exb/sdrangel #1661: Crash when transmitting in SSB

sh123, 2023-04-16:
RPI4, master version, when using SSB transmit plugin to transmit from microphone through pulseaudio rtp module, after a while application is freezing and then crashes with `corrupted double-linked list`
```
2023-04-16 11:22:11.694 (D) CWKeyer::applySettings:   m_dashKey:  Qt::Key_Minus  m_dashKeyModifiers:  QFlags<Qt::KeyboardModifier>(NoModifier)  m_dotKey:  Qt::Key_Period  m_dotKeyModifiers:  QFlags<Qt::KeyboardModifier>(NoModifier)  m_keyboardIambic:  true  m_loop:  false  m_mode:  0  m_sampleRate:  48000  m_text:  ""  m_wpm:  13
2023-04-16 11:22:11.694 (D) CWKeyer::handleMessage: MsgConfigureCWKeyer
2023-04-16 11:22:11.694 (D) CWKeyer::applySettings:   m_dashKey:  Qt::Key_Minus  m_dashKeyModifiers:  QFlags<Qt::KeyboardModifier>(NoModifier)  m_dotKey:  Qt::Key_Period  m_dotKeyModifiers:  QFlags<Qt::KeyboardModifier>(NoModifier)  m_keyboardIambic:  true  m_loop:  false  m_mode:  0  m_sampleRate:  48000  m_text:  ""  m_wpm:  13
2023-04-16 11:22:14.829 (D) SSBMod::handleMessage: MsgConfigureSSBMod
2023-04-16 11:22:14.829 (D) SSBModBaseband::handleMessage: MsgConfigureSSBModBaseband
2023-04-16 11:22:14.829 (D) AudioDeviceManager::addAudioSource: 11: 0x5580cf8fe8
2023-04-16 11:22:15.357 (W) AudioInputDevice::start: using audio device #11: rtpfrompc.monitor
2023-04-16 11:22:15.357 (I) AudioInputDevice::start: audio format OK
corrupted double-linked list
Aborted (core dumped)
```

```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000007f99d8aaa0 in __GI_abort () at abort.c:79
#2  0x0000007f99dd7e80 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7f99e9b648 "%s\n")
    at ../sysdeps/posix/libc_fatal.c:155
#3  0x0000007f99ddf3dc in malloc_printerr (str=str@entry=0x7f99e97040 "corrupted double-linked list") at malloc.c:5347
#4  0x0000007f99ddffb4 in unlink_chunk (p=p@entry=0x7ee84d91c0, av=0x7ee8000020) at malloc.c:1460
#5  0x0000007f99de27a0 in _int_malloc (av=av@entry=0x7ee8000020, bytes=bytes@entry=1637) at malloc.c:4041
#6  0x0000007f99de3838 in __GI___libc_malloc (bytes=1637) at malloc.c:3066
#7  0x0000007f9a1b636c in QArrayData::allocate(unsigned long, unsigned long, unsigned long, QFlags<QArrayData::AllocationOption>) () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#8  0x0000007f9a1f5a48 in QByteArray::QByteArray(int, Qt::Initialization) () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#9  0x0000007f883bf65c in  () at /usr/lib/aarch64-linux-gnu/qt5/plugins/audio/libqtmedia_pulse.so
#10 0x0000007f883bfad4 in  () at /usr/lib/aarch64-linux-gnu/qt5/plugins/audio/libqtmedia_pulse.so
#11 0x0000007f9a3d14d4 in  () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#12 0x0000007f9a3d5cc0 in QTimer::timeout(QTimer::QPrivateSignal) () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#13 0x0000007f9a3c66e8 in QObject::event(QEvent*) () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#14 0x0000007f9ae5da80 in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
    at /lib/aarch64-linux-gnu/libQt5Widgets.so.5
#15 0x0000007f9a394460 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#16 0x0000007f9a3f5408 in QTimerInfoList::activateTimers() () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#17 0x0000007f9a3f5dac in  () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#18 0x0000007f98779bc8 in g_main_context_dispatch () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#19 0x0000007f98779e5c in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#20 0x0000007f98779f24 in g_main_context_iteration () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#21 0x0000007f9a3f6154 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /lib/aarch64-linux-gnu/libQt5Core.so.5
#22 0x0000007f9a3928ac in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /lib/aarch64-linux-gnu/libQt5Core.so.5
#23 0x0000007f9a1a79ec in QThread::exec() () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#24 0x0000007f9a1a8e6c in  () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#25 0x0000007f99177648 in start_thread (arg=0x7efca3d980) at pthread_create.c:477
#26 0x0000007f99e3cc1c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
(gdb) 
```

closed 2023-04-20 by sh123

f4exb, 2023-04-21:
Well I also have quantity of issues related to `QRingBuffer::read` mostly some kind of infinite loop but sometimes similar memory issues so there could be something fishy here. Very difficult to track since there is no SDRangel code directly involved similarly to the stack trace shown above. Possibly a Qt issue. This is random and in my case I have SSB demods (not mods) in my setup.

sh123, 2023-04-21:
@f4exb , I cannot reproduce this issue with RX, only when TX, while it might be issue with QT/pulse corrupting memory, but under valgrind there is a number of invalid memory reads in other places, for example
```
023-04-21 12:01:13.334 (D) GLShaderSpectrogram::initGrid: requested:  256  actual:  256
2023-04-21 12:01:15.765 (W) SampleSourceFifo::write: underrun (write too slow) using 9600 old samples
==19650== Thread 21 QThread:
==19650== Invalid read of size 8
==19650==    at 0x4851268: __GI_memmove (vg_replace_strmem.c:1271)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:426)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:472)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:506)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:513)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:569)
==19650==    by 0x49BCEA37: SSBModSource::pullAudio(unsigned int) (ssbmodsource.cpp:156)
==19650==    by 0x49BC9B33: SSBModBaseband::processFifo(std::vector<Sample, std::allocator<Sample> >&, unsigned int, unsigned int) (ssbmodbaseband.cpp:128)
==19650==    by 0x49BC9C2F: SSBModBaseband::handleData() (ssbmodbaseband.cpp:116)
==19650==    by 0x653E7D3: QObject::event(QEvent*) (in /usr/lib/aarch64-linux-gnu/libQt5Core.so.5.15.2)
==19650==    by 0x568DA7F: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /usr/lib/aarch64-linux-gnu/libQt5Widgets.so.5.15.2)
```

```
19650==  Address 0x1a7f6360 is 0 bytes after a block of size 96,000 alloc'd
==19650==    at 0x484A484: operator new(unsigned long) (vg_replace_malloc.c:342)
==19650==    by 0x49BCE1F3: std::vector<AudioSample, std::allocator<AudioSample> >::_M_default_append(unsigned long) (new_allocator.h:115)
==19650==    by 0x49BD1BB3: UnknownInlinedFun (stl_vector.h:940)
==19650==    by 0x49BD1BB3: SSBModSource::SSBModSource() (ssbmodsource.cpp:54)
==19650==    by 0x49BCD3FF: SSBModBaseband::SSBModBaseband() (ssbmodbaseband.cpp:29)
==19650==    by 0x49BCD803: SSBMod::SSBMod(DeviceAPI*) (ssbmod.cpp:67)
==19650==    by 0x49BD457B: non-virtual thunk to SSBModPlugin::createTxChannel(DeviceAPI*, BasebandSampleSource**, ChannelAPI**) const (ssbmodplugin.cpp:61)
==19650==    by 0x49DF193: DeviceUISet::loadTxChannelSettings(Preset const*, PluginAPI*, QList<Workspace*>*, Workspace*) (deviceuiset.cpp:434)
==19650==    by 0x4907CD3: MainWindow::loadConfiguration(Configuration const*, bool) (mainwindow.cpp:1484)
```

```
==19650== Invalid read of size 8
==19650==    at 0x4851278: __GI_memmove (vg_replace_strmem.c:1271)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:426)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:472)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:506)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:513)
==19650==    by 0x49BCEA37: UnknownInlinedFun (stl_algobase.h:569)
==19650==    by 0x49BCEA37: SSBModSource::pullAudio(unsigned int) (ssbmodsource.cpp:156)
==19650==    by 0x49BC9B33: SSBModBaseband::processFifo(std::vector<Sample, std::allocator<Sample> >&, unsigned int, unsigned int) (ssbmodbaseband.cpp:128)
==19650==    by 0x49BC9C2F: SSBModBaseband::handleData() (ssbmodbaseband.cpp:116)
==19650==    by 0x653E7D3: QObject::event(QEvent*) (in /usr/lib/aarch64-linux-gnu/libQt5Core.so.5.15.2)
```

sh123, 2023-04-22:
@f4exb , unfortunately this chance did not fix issue for me as it is only for RX, but it also broke RX as receive stops working after clicking on STOP->START for the selected SDR receiver, it starts producing next lines, only app restart helps.
```
2023-04-22 22:46:53.880 (D) SSBDemodSink::processOneSample: 0/16384 samples written
2023-04-22 22:46:54.239 (C) AudioFifo::write: () overflow 16384 samples
2023-04-22 22:46:54.239 (D) SSBDemodSink::processOneSample: 0/16384 samples written
2023-04-22 22:46:54.565 (C) AudioFifo::write: () overflow 16384 samples
2023-04-22 22:46:54.565 (D) SSBDemodSink::processOneSample: 0/16384 samples written
2023-04-22 22:46:54.917 (C) AudioFifo::write: () overflow 16384 samples
2023-04-22 22:46:54.917 (D) SSBDemodSink::processOneSample: 0/16384 samples written
2023-04-22 22:46:55.245 (C) AudioFifo::write: () overflow 16384 samples
2023-04-22 22:46:55.245 (D) SSBDemodSink::processOneSample: 0/16384 samples written
```

f4exb, 2023-04-24:
I don't think this is related to the change. If you look at the code what it does is just a sanity check:
```c++
        if (m_audioBufferFill >= m_audioBuffer.size())
        {
            if (m_audioBufferFill > m_audioBuffer.size()) {
                qDebug("SSBDemodSink::processOneSample: dropping %d samples", (int) m_audioBufferFill - (int) m_audioBuffer.size());
            }
```
Normally `m_audioBufferFill ` should never exceed ` m_audioBuffer.size()` but since the precaution was made to do a `>=` comparison instead of `==` it remains a half-baked solution if you don't check for `>` later. This is just what this change does.

sh123, 2023-04-24:
Yes, sorry, There must be something wrong with pulse/qt5 in RPIs bullsye distro , restarted RPI and stop->play SDR is able to start audio again.

But with TX app just completely freezes now after about 10 seconds of transmission.

This setup worked for me in the past with previous RPI distro and older SDRAngel version (about 1 year old). Must be some regression in pulseaudio... will re-try when they will release updates.

srcejon, 2023-04-25:
Yep, I seem to be seeing the same problem as @sh123 after this change.

srcejon, 2023-04-25:
Tried SSB, VOR and DAB and all stutter and GUI becomes unresponsive. Demods without audio are OK.

sh123, 2023-04-25:
Interesting, that this issue does not happen when there is only one way RTP stream from PC to RPI running SDRAngel (only tx-ing through ssb, not rx-ing anything). When there are two streams from PC to RPI and from RPI to PC then this issue happens.

f4exb, 2023-04-25:
I am sorry that this part (at least) is so badly written that this simple change causes issues. I will fallback but this does not encourages me to continue with development of this software.

sh123, 2023-04-25:
@f4exb, we all like this software, code is readable, no problems, made a number of QSOs over QO-100 by using it and remote outdoors station... just something got broken in qt/pulse after RPI upgrade :/

f4exb, 2023-04-25:
OK glad to hear that. I hate it when it goes brittle for no apparent reason.

sh123, 2023-04-27:
Maybe somebody will find it useful, this workaround does not trigger crash:
- Use RTP stream from PC to RPI
```
pactl load-module module-native-protocol-tcp auth-ip-acl='127.0.0.1;192.168.0.0/16' auth-anonymous=1
pactl load-module module-rtp-send source=<source mic device> destination_ip=<rpi ip> rate=8000 channels=1
```
- Use TCP from RPI to PC, it avoids creating null sink on RPI side
```
pactl load-module module-tunnel-sink-new server=<pc ip> sink=<pc playback device> rate=8000 channels=1
pactl load-module module-null-sink sink_name=rtpfrompc sink_properties="device.description='RPI<-PC'"
pactl load-module module-rtp-recv sink=rtpfrompc sap_address=0.0.0.0 latency_msec=100
```