f4exb/sdrangel #1662: Unable to build properly SDRangel on ODROID N2+ - arm64

mathis-lambert, 2023-04-17:
### **Hi everyone,**

###  Unable to build properly SDRangel on ODROID N2+ - arm64

I'm a student working on  **analyzing private 5G network waves**,
My hardware is an **ODROID N2plus with Ubuntu Mate 22.04 (Jammy)**
based on the **arm64 architecture.**

I'm working with the LimeSDR Mini and by the fact that no arm64 version has been officially released (even after trying community builds that don't even work) I decided to build my own version internally on my own platform.

After having CAREFULLY read/followed the doc _build_from_source_
SDR Angel launches SLOWLY but properly but when i want to add external devices,
there's absolutely no devices recognized.

I've of course included LimeSDR hardware requirements mentioned in the documentation but there's nothing to do .... 

### What i've tested

> I've checked if the Limesuite was included : **it does**

> i've checked if lime and the odroid detect the LimeSDR with `lsusb` or `LimeUtil --find` : **it does**

> I've rebuilt two times the software : nothing changed

Thank you in advance for any help you can provide!

Kind regards :)

srcejon, 2023-04-17:
Most up to date compilation instructions are here: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

Might be useful to post your log file.

mathis-lambert, 2023-04-17:
@srcejon 

This is the doc i followed.
i've done the whole process step by step,
The build executed properly and finished with no error.

i don't know what else i can do ....

srcejon, 2023-04-17:
Post your log file...

mathis-lambert, 2023-04-18:
@srcejon 

Here is the log file when i plug the lime in, and open sdrangel

`Apr 18 06:31:20 odroid kernel: [  363.157994@4] usb 2-1.4: New USB device found, idVendor=0403, idProduct=601f
Apr 18 06:31:20 odroid kernel: [  363.157996@4] usb 2-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 18 06:31:20 odroid kernel: [  363.157998@4] usb 2-1.4: Product: LimeSDR Mini
Apr 18 06:31:20 odroid kernel: [  363.158000@4] usb 2-1.4: Manufacturer: Lime Micro
Apr 18 06:31:20 odroid kernel: [  363.158001@4] usb 2-1.4: SerialNumber: 1D42564C6B4307
Apr 18 06:31:20 odroid kernel: [  363.187867@4] usb 2-1.4: Unsupported device
Apr 18 06:31:20 odroid kernel: [  363.187956@4] usb 2-1.4: Unsupported device
Apr 18 06:31:20 odroid mtp-probe: checking bus 2, device 3: "/sys/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1.4"
Apr 18 06:31:20 odroid mtp-probe: bus: 2, device: 3 was not an MTP device
Apr 18 06:31:24 odroid systemd[3436]: Started Application launched by gnome-shell.
Apr 18 06:31:25 odroid dbus-daemon[3468]: [session uid=1000 pid=3468] Activating via systemd: service name='org.gnome.Terminal' unit='gnome-terminal-server.service' requested by ':1.110' (uid=1000 pid=5221 comm="/usr/bin/gnome-terminal.real ")
Apr 18 06:31:25 odroid systemd[3436]: Created slice Slice /app/org.gnome.Terminal.
Apr 18 06:31:25 odroid systemd[3436]: Starting GNOME Terminal Server...
Apr 18 06:31:25 odroid dbus-daemon[3468]: [session uid=1000 pid=3468] Successfully activated service 'org.gnome.Terminal'
Apr 18 06:31:25 odroid systemd[3436]: Started GNOME Terminal Server.
Apr 18 06:31:25 odroid systemd[3436]: Started VTE child process 5250 launched by gnome-terminal-server process 5228.
Apr 18 06:31:44 odroid systemd[3436]: Started Application launched by gnome-shell.
Apr 18 06:31:47 odroid rtkit-daemon[3462]: Supervising 8 threads of 3 processes of 1 users.
Apr 18 06:31:47 odroid rtkit-daemon[3462]: message repeated 5 times: [ Supervising 8 threads of 3 processes of 1 users.]
Apr 18 06:31:47 odroid rtkit-daemon[3462]: Successfully made thread 5421 of process 5265 owned by '1000' RT at priority 10.
Apr 18 06:31:47 odroid rtkit-daemon[3462]: Supervising 9 threads of 4 processes of 1 users.
Apr 18 06:31:48 odroid rtkit-daemon[3462]: message repeated 4 times: [ Supervising 9 threads of 4 processes of 1 users.]
Apr 18 06:31:50 odroid firefox.desktop[5265]: [ERROR glean_core] Error setting metrics feature config: Json(Error("EOF while parsing a value", line: 1, column: 0))
Apr 18 06:31:50 odroid rtkit-daemon[3462]: Supervising 9 threads of 4 processes of 1 users.
Apr 18 06:31:50 odroid rtkit-daemon[3462]: message repeated 5 times: [ Supervising 9 threads of 4 processes of 1 users.]
Apr 18 06:31:55 odroid firefox.desktop[5265]: libEGL warning: DRI2: failed to authenticate
Apr 18 06:31:55 odroid firefox.desktop[5265]: ATTENTION: default value of option mesa_glthread overridden by environment.
Apr 18 06:32:04 odroid rtkit-daemon[3462]: Supervising 9 threads of 4 processes of 1 users.
Apr 18 06:32:05 odroid rtkit-daemon[3462]: message repeated 5 times: [ Supervising 9 threads of 4 processes of 1 users.]
Apr 18 06:32:05 odroid rtkit-daemon[3462]: Successfully made thread 5681 of process 5535 owned by '1000' RT at priority 10.
Apr 18 06:32:05 odroid rtkit-daemon[3462]: Supervising 10 threads of 5 processes of 1 users.
Apr 18 06:32:06 odroid rtkit-daemon[3462]: message repeated 2 times: [ Supervising 10 threads of 5 processes of 1 users.]
Apr 18 06:32:54 odroid rtkit-daemon[3462]: Supervising 9 threads of 4 processes of 1 users.
Apr 18 06:33:03 odroid rtkit-daemon[3462]: message repeated 7 times: [ Supervising 9 threads of 4 processes of 1 users.]
Apr 18 06:33:03 odroid anacron[4706]: Job `cron.daily' started
Apr 18 06:33:03 odroid anacron[5880]: Updated timestamp for job `cron.daily' to 2023-04-18
Apr 18 06:33:04 odroid cracklib: no dictionary update necessary.
Apr 18 06:33:04 odroid anacron[4706]: Job `cron.daily' terminated
Apr 18 06:33:04 odroid anacron[4706]: Normal exit (1 job run)
Apr 18 06:33:04 odroid systemd[1]: anacron.service: Deactivated successfully.
Apr 18 06:33:31 odroid dbus-daemon[3468]: [session uid=1000 pid=3468] Activating via systemd: service name='org.freedesktop.Tracker3.Miner.Extract' unit='tracker-extract-3.service' requested by ':1.8' (uid=1000 pid=3537 comm="/usr/libexec/tracker-miner-fs-3 ")
Apr 18 06:33:31 odroid systemd[3436]: Starting Tracker metadata extractor...
Apr 18 06:33:31 odroid dbus-daemon[3468]: [session uid=1000 pid=3468] Successfully activated service 'org.freedesktop.Tracker3.Miner.Extract'
Apr 18 06:33:31 odroid systemd[3436]: Started Tracker metadata extractor.
Apr 18 06:34:11 odroid dbus-daemon[3205]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.83' (uid=1000 pid=4284 comm="/usr/bin/nautilus --gapplication-service ")
Apr 18 06:34:11 odroid systemd[1]: Starting Hostname Service...
Apr 18 06:34:11 odroid dbus-daemon[3205]: [system] Successfully activated service 'org.freedesktop.hostname1'
Apr 18 06:34:11 odroid systemd[1]: Started Hostname Service.
Apr 18 06:34:19 odroid nautilus[4284]: Called "net usershare info" but it failed: 'net usershare' returned error 255: net usershare: cannot open usershare directory /var/lib/samba/usershares. Error No such file or directory#012Please ask your system administrator to enable user sharing.
[... cut ...]
Apr 18 06:39:02 odroid org.gnome.Nautilus[6134]: 2023-04-18 06:39:02.524 (D) MainWindow::MainWindow: end
Apr 18 06:39:23 odroid dbus-daemon[3468]: [session uid=1000 pid=3468] Activating service name='org.gnome.gedit' requested by ':1.95' (uid=1000 pid=4284 comm="/usr/bin/nautilus --gapplication-service ")
Apr 18 06:39:23 odroid dbus-daemon[3468]: [session uid=1000 pid=3468] Successfully activated service 'org.gnome.gedit'
Apr 18 06:39:23 odroid gedit[6308]: The specified location is not mounted`

srcejon, 2023-04-18:
Many plugins appear not to be loaded from that log file. You should have:

    PluginManager::loadPluginsDir: fileName: inputlimesdr.dll
    PluginManager::loadPluginsDir: loaded plugin inputlimesdr.dll

To use the Lime.

Next step is to check whether they have been built or not. Then if not, why not. You'll need to review your build logs.

mathis-lambert, 2023-04-18:
@srcejon 

I don't exactly know how to check whether it has been built or not, but by the fact that a folder named limesdrinput inside /build/plugins it might have been.

In the doc you provide, i followed each steps which include LimeSuite even with Soapy

srcejon, 2023-04-18:
What's the output of:

    ls /opt/install/sdrangel/lib/sdrangel/plugins

And:

    ls /opt/build/sdrangel/build/plugins/samplesource/limesdrinput/CMakeFiles/inputlimesdr.dir

You could also post CMakeCache.txt from your build directory.

mathis-lambert, 2023-04-18:
@srcejon 

[image]

And the CMakeCache.txt

[CMakeCache.txt](https://github.com/f4exb/sdrangel/files/11262124/CMakeCache.txt)

srcejon, 2023-04-18:
Did you run make install?

Are you running from the build directory rather than the install directory?

mathis-lambert, 2023-04-18:
@srcejon 

Yes ! I ran `make -j $(nproc) install` in `/opt/build/sdrangel/build`

And i'm running the software from this folder

srcejon, 2023-04-18:
Did you run cmake with the option -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel  ? That specifies the directory where it should be installed to - and the directory you should run it from.

But the 'cannot access' message above suggests you didn't.

mathis-lambert, 2023-04-18:
`cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DMIRISDR_DIR=/opt/install/libmirisdr \
-DAIRSPY_DIR=/opt/install/libairspy \
-DAIRSPYHF_DIR=/opt/install/libairspyhf \
-DBLADERF_DIR=/opt/install/libbladeRF \
-DHACKRF_DIR=/opt/install/libhackrf \
-DRTLSDR_DIR=/opt/install/librtlsdr \
-DLIMESUITE_DIR=/opt/install/LimeSuite \
-DIIO_DIR=/opt/install/libiio \
-DPERSEUS_DIR=/opt/install/libperseus \
-DXTRX_DIR=/opt/install/xtrx-images \
-DSOAPYSDR_DIR=/opt/install/SoapySDR \
-DUHD_DIR=/opt/install/uhd \
-DAPT_DIR=/opt/install/aptdec \
-DCM256CC_DIR=/opt/install/cm256cc \
-DDSDCC_DIR=/opt/install/dsdcc \
-DSERIALDV_DIR=/opt/install/serialdv \
-DMBE_DIR=/opt/install/mbelib \
-DCODEC2_DIR=/opt/install/codec2 \
-DSGP4_DIR=/opt/install/sgp4 \
-DLIBSIGMF_DIR=/opt/install/libsigmf \
-DDAB_DIR=/opt/install/libdab \
-DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..`

I ran it with all these options. Wen can see the line you mentionned before

As i said before, i have carefully read the doc compile_from_source

srcejon, 2023-04-18:
What was output when you ran make install?

Perhaps it couldn't install due to permissions.

mathis-lambert, 2023-04-18:
@srcejon 

You can find here what i got from the command `make install`

[build_log.txt](https://github.com/f4exb/sdrangel/files/11263709/build_log.txt)

To answer your last message, yes i did the chown command...

srcejon, 2023-04-18:
The problem is the source is not being compiled fully:

    c++: fatal error: Killed signal terminated program cc1plus
    compilation terminated.
    make[2]: *** [plugins/channelrx/demodadsb/CMakeFiles/demodadsb.dir/build.make:2432: 
    plugins/channelrx/demodadsb/CMakeFiles/demodadsb.dir/demodadsb_autogen/EWIEGA46WW/qrc_airlinelogos.cpp.o] 
    Error 1
    make[1]: *** [CMakeFiles/Makefile2:5791: plugins/channelrx/demodadsb/CMakeFiles/demodadsb.dir/all] Error 2
    make[1]: *** Waiting for unfinished jobs....

Possibly because you are running out of memory, if you're just using a small embedded computer.

It might help if you compile without debug info - but this is just a guess. Change:

    -DCMAKE_BUILD_TYPE=RelWithDebInfo 

To 

    -DCMAKE_BUILD_TYPE=Release

srcejon, 2023-04-18:
Or, don't do a parallel build. Change:

     make -j $(nproc) install

To just

    make install

mathis-lambert, 2023-04-19:
Hi @srcejon 

I ran a new build but it crashed again and again ...

[error_build.txt](https://github.com/f4exb/sdrangel/files/11270643/error_build.txt)

It's kind of crazy because sometimes it works and sometimes not

srcejon, 2023-04-19:
Well, that confirms the problem:

    cc1plus: out of memory allocating 268435408 bytes after a total of 161779712 bytes

If you don't want to use ADS-B demod, you could temporarily disable that, by adding the following to cmake command

   -DENABLE_CHANNELRX_DEMODADSB=OFF

You may need to disable the Star Tracker as well:

   -DENABLE_FEATURE_STARTRACKER=OFF

mathis-lambert, 2023-04-19:
@srcejon 

I've got a new FATAL ERROR sayin :

```
> cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
> c++: fatal error: Killed signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [plugins/feature/map/CMakeFiles/map.dir/build.make:701: plugins/feature/map/CMakeFiles/map.dir/map_autogen/EWIEGA46WW/qrc_cesium.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:8809: plugins/feature/map/CMakeFiles/map.dir/all] Error 2
> make: *** [Makefile:156: all] Error 2
```

If you want the output :
[error_build.txt](https://github.com/f4exb/sdrangel/files/11272964/error_build.txt)
[output.txt](https://github.com/f4exb/sdrangel/files/11272965/output.txt)

I ran the executable from this build and no usb again

This is what syslog says when i click 'update usb devices' on SDRAngel

 
> Apr 19 12:25:14 odroid dbus-daemon[3461]: [session uid=1000 pid=3461] Activating service name='org.gnome.gedit' requested by ':1.144' (uid=1000 pid=17867 comm="/usr/bin/nautilus --gapplication-service ")
> Apr 19 12:25:14 odroid dbus-daemon[3461]: [session uid=1000 pid=3461] Successfully activated service 'org.gnome.gedit'

srcejon, 2023-04-19:
You can disable the map with: -DENABLE_FEATURE_MAP=OFF

> I ran the executable from this build and no usb again

Until you have a build that says 100% complete, there's no point it trying.

mathis-lambert, 2023-04-19:
@srcejon 

Do the final percentage change depending which plugins i include ?
Because i already ran a complete build (no fatal error but 2 errors) and the ending percentage was about 71%

mathis-lambert, 2023-04-19:
@srcejon 

I now, run this command :

> cd /opt/build/sdrangel/build/
make clean
cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
-DCMAKE_BUILD_TYPE=Release \
-DMIRISDR_DIR=/opt/install/libmirisdr \
-DAIRSPY_DIR=/opt/install/libairspy \
-DAIRSPYHF_DIR=/opt/install/libairspyhf \
-DBLADERF_DIR=/opt/install/libbladeRF \
-DHACKRF_DIR=/opt/install/libhackrf \
-DRTLSDR_DIR=/opt/install/librtlsdr \
-DLIMESUITE_DIR=/opt/install/LimeSuite \
-DIIO_DIR=/opt/install/libiio \
-DPERSEUS_DIR=/opt/install/libperseus \
-DXTRX_DIR=/opt/install/xtrx-images \
-DSOAPYSDR_DIR=/opt/install/SoapySDR \
-DUHD_DIR=/opt/install/uhd \
-DAPT_DIR=/opt/install/aptdec \
-DCM256CC_DIR=/opt/install/cm256cc \
-DDSDCC_DIR=/opt/install/dsdcc \
-DSERIALDV_DIR=/opt/install/serialdv \
-DMBE_DIR=/opt/install/mbelib \
-DCODEC2_DIR=/opt/install/codec2 \
-DSGP4_DIR=/opt/install/sgp4 \
-DLIBSIGMF_DIR=/opt/install/libsigmf \
-DDAB_DIR=/opt/install/libdab \
-DENABLE_CHANNELRX_DEMODADSB=OFF \
-DENABLE_FEATURE_STARTRACKER=OFF \
-DENABLE_FEATURE_MAP=OFF \
-DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
make install > /home/odroid/Desktop/output.txt

mathis-lambert, 2023-04-20:
It works !!!

[image]

Thank you very much, I blame chance, I think the CPU was ok but it's a bit when it wants ...

For those having the same issue, to make it working, my build indeed reached 100%
You can find below my last output files (with deprecations alerts and progress)

[deprecations_warns.txt](https://github.com/f4exb/sdrangel/files/11283464/deprecations_warns.txt)
[progress.txt](https://github.com/f4exb/sdrangel/files/11283465/progress.txt)

Just to sum up what i needed to do :

Disable some plugins to free up some memory, so after having followed [compile_from_source](https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux)

So i finally ran this command :

> cd /opt/build/sdrangel/build/
> make clean
> cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
> -DCMAKE_BUILD_TYPE=Release \
> -DMIRISDR_DIR=/opt/install/libmirisdr \ [...]

After waiting during many hours (5-6), i'm able to find the executable in /opt/install/sdrangel

/!\ /!\ there is still an issue i was not able to handle, the allocating FFTs process at the start of SDRAngel is soooo long and i don't know why ...

Thanks again to @srcejon

srcejon, 2023-04-20:
> here is still an issue i was not able to handle, the allocating FFTs process at the start of SDRAngel is soooo long 

Preferences > FFT and generate the Wisdom file.

mathis-lambert, 2023-04-20:
Thank you again