f4exb/sdrangel #1672: Accessibility of SDRangel for screen reader users

arfy8820, 2023-04-28:
Hello. First, I'd like to say I'm fairly new to the world of SdR, but loving it so far. as a long-time radio enthusiast.
I've also been on the hunt for an accessible and usable SDR software sollution that's updated, and this is the most promising so far on all counts. However, I've noticed some difficulties, but first, I have to congratulate you on the following.
* Your use of standard QT controls that work well with screen readers out of the box, especially under windows.
* the general layout of the program seems easy to understand, especially once I worked out how modular it seems you've intended the program to be.

Now, for the improvements that could be made.
* More keyboard shortcuts included! To help navigation around the program, it would be good to have hotkeys to switch among workspaces, devices, and channels. Ctrl-tab does work somewhat, but it can be tricky to keep up with where the focus is and what element is being worked with.
* The biggy for me, it seems hard to adjust the frequency using just the keyboard, unless I haven't worked out something. This is the one control that doesn't seem to be a standard slider or input field, etc. I can tab to the control that, with my screen access sollution says something like "center frequency" but it seems I can't actually adjust it. I've tried all combos of the arrow keys, typing, or just hitting space/enter to try get something to happen.
* Related. what's the procedure for scanning through an IQ recording? From my limited tests I've been able to carry out, you can't adjust the center frequency, instead, you adjust the offset frequency in the demodulator. For example, FM broadcast. Is that correct?

In summary, the accessibility of the program for blind/visually impaired users using a screen reader is very promising, although some tweaks could be made. My knowledge of c++ is limited, but I can hopefully offer pointers on how to make the program even more accessible using the existing QT framework, or, create an alternative interface using the web API, which is also very impressive!

srcejon, 2023-04-28:
> The biggy for me, it seems hard to adjust the frequency using just the keyboard, unless I haven't worked out something. This is the one control that doesn't seem to be a standard slider or input [...]

Which SDR device is this with and what OS? I just tried with the RTL SDR device on Windows, and I can tab to the centre frequency control and enter a frequency using digit keys OK (and use left/right cursor keys to move which digit to change).

> From my limited tests I've been able to carry out, you can't adjust the center frequency, instead, you adjust the offset frequency in the demodulator.

Correct.

arfy8820, 2023-04-28:
Thanks for the reply. I'll try again with a radio live on air, this was just testing with IQ files, as I'm waiting for an antenna to come in to put my SDR on air. I'll try adjusting the frequency with the demodulator, even so, the screen reader doesn't announce which digit is being adjusted, as far as I know. Will test more and let you know.

arfy8820, 2023-04-28:
Update. this works a treat! At least in terms of adjusting frequency while playing back recorded IQ files. However, as previously noted, the screen reader gives no indication of which digit is being adjusted, or even what the frequency is when you land on the control. This seems to apply no matter where this control shows up. Center frequency, offset frequency, or in the spectrum markers dialog.

srcejon, 2023-04-28:
With this patch, https://github.com/f4exb/sdrangel/pull/1675, I've added some accessibility support for the ValueDial widgets (used for centre frequency and frequency offset), so screen readers should be able to say what the frequency is when the widget gets focus and when the value is adjusted. Tested with Windows Narrator.

Note that while testing this, I noticed that QDials don't seem to work (no value is read) - this appears to be a Qt bug, rather than SDRangel: https://bugreports.qt.io/browse/QTBUG-86038

arfy8820, 2023-04-28:
Thank you! First, for caring and taking the time to patch this (I'm setting up dev tools now to test this), and second, for knowing what needed to be looked into. I only had a vague idea of knowing that something would need to be done to tell QT about the custom control for the accessibility layer.
I'll close this issue and report further things on the new thread you've created for this patch.

arfy8820, 2023-04-28:
Thanks for working on this! For taking the time and caring, and also knowing where to look to make changes at the accessibility level. Now installing dev tools and will test the patch.

closed 2023-04-28 by arfy8820

arfy8820, 2023-04-28:
Ok, reopened this issue for further discussion. I'm still a little noob at github sometimes, even though I've got my own repositories on here! dough!

srcejon, 2023-04-28:
> and also knowing where to look to make changes at the accessibility level.

Had no idea to be honest!

One thing I did discover while looking at this, that you might be able to help improve, is the name & description the screen reader gives when any widget gets focus. All of the GUI's have a .ui file that you can load in Qt Creator. When a widget is selected, two of the properties listed are:

    accessibleName
    accessibleDescription

These are all currently empty. It might help filling some of these in, to something you find more useful, as often the tool tips are used instead, which can be a little verbose (Great for tool tips, but possibly not ideal for this).

arfy8820, 2023-04-28:
Good idea! Although a lot are just fine as is, but this is good to know. Related, the latest QT installer is wanting to give me version 6.5, is this ok, or should I try track down the older 5.15.2 version you mention? Might be worth seeing if accessibility bugs like the one you found for QDial is fixed. although I know going from one major version to another can be a pain in the ...

srcejon, 2023-04-28:
I wouldn't recommend Qt 6.5 for now - it doesn't have all the functionality of 5.x. You should be able to install 5.12.2 from the same installer.

arfy8820, 2023-04-28:
Got it! Had to poke around in the custom settings. Let the show continue. :)

arfy8820, 2023-05-01:
Status update. I'm impressed! The accessibility patch is working as advertised, I can navigate and read the frequency values as they change. The next tweak to make is to announce the selected digit (or plus/minus sign for ValueDialZ widgets), as the left and right cursor keys are used. I'll also give QT Creator a look, but often these large IDE type programs aren't very accessible for us. Another simple tweak that can be made for a lot of controls is to add shortcut keys. For example, the "open" button in the file device window could have O as its shortcut. You can simply add an & sign in front of the letter that you want for the shortcut on the button's text.

gsasa, 2025-09-19:
Specifically, when you first start the application after installation, I chose the configuration for DAB. I chose a Channel 11b and I'm stuck here. I guess it should be somewhere a button to activate, but I'm not doing anything that would assocate me to that.

srcejon, 2025-09-19:
> Specifically, when you first start the application after installation, I chose the configuration for DAB. I chose a Channel 11b and I'm stuck here. I guess it should be somewhere a button to activate, but I'm not doing anything that would assocate me to that.

Ok, I think I understand. When you open the DAB configuration, there is a tab view with 3 tabs. One for the RTL SDR Device that has the start button, one for the spectrum and the third showing the DAB Demodulator. You can switch between tabs by pressing CTRL-TAB, but this probably isn't obvious without being able to see them, as the narrator doesn't appear to mention them.

It appears that repeatedly pressing tab doesn't take you to the toolbar, and we indeed don't have keyboard shortcuts, so these do seem in-accessible. I can add the following shortcuts:

CTRL-P - Configuration Presets
CTRL-S - Start/stop all devices
CTRL-R - Add RX device
CTRL-T - Add TX device
CTRL-M - Add MIMO device
CTRL-A - Add feature (CTRL-F is already used for File menu)
CTRL-SHIFT-P - Feature presets

CTRL-SHIFT-C - Cascade windows
CTRL-SHIFT-T - Tile windows
CTRL-SHIFT-V - Stack windows vertically
CTRL-SHIFT-S - Stack windows in columns
CTRL-SHIFT-B - Display windows in tabs

@f4exb - Please let me know if you think any of these will clash with anything else. I don't really use the keyboard.