f4exb/sdrangel #1674: Satellite Radio Control Preset

abraxasneo, 2023-04-28:
I have version 17.13.0 and am working with RSPDuo sdr. 
The problem is that the frequency presets for the selected satellites are not displayed.
 I am trying with different satellites but to no avail. 
[image]

abraxasneo, 2023-04-28:
There are no presets there when this dialog is opened. That is why my question. Maybe it's just my simple mistake, but I've run out of ideas
[image]

srcejon, 2023-04-28:
Yes, this displays Device Set presets - which you need to create yourself.

Close that dialog and setup SDRangel how you need it for that particular satellite - then save a Device Set preset using the Star button in the Device window. Then that preset will appear in the combo box you've highlighed.

abraxasneo, 2023-04-28:
OK :) Thank you!
 I naively thought that the program automatically matches frequencies and demodulators to the appropriate satellite modes....