f4exb/sdrangel #1677: Error while compiling on windows 11 64-bit

arfy8820, 2023-04-29:
I'm trying to compile the latest sources from git, and am able to complete all instructions up to ninja. However, ninja stops with.
```
ninja: build stopped: subcommand failed.
```
The output leading up to this was.
```
testThreads.c
        link.exe /nologo /VERSION:2.10 /LIBPATH:bin.msvc /LIBPATH:.\lib  /OUT:bin.msvc\testThreads.exe libxml2.lib  wsock32.lib ws2_32.lib kernel32.lib int.utils.msvc\testThreads.obj
```
Where should I go to try get further logs/info about what might be happening?
My system config is.
* Windows 11 22H2, 64-bit.
* Visual C++ 2022.
* QT 5.15.2
* Boost 1.82.0
* cmake 3.24.2
* Ninja 1.10.2

srcejon, 2023-04-29:
Can you post more of the output please. The above doesn't show the compiler error.

arfy8820, 2023-04-29:
[cmake output.txt](https://github.com/f4exb/sdrangel/files/11358512/cmake.output.txt)
Here's my complete console log of the build session.

srcejon, 2023-04-29:
The actual error seems to be:

    Windows Subsystem for Linux has no installed distributions.   
    Distributions can be installed by visiting the Microsoft Store:
    https://aka.ms/wslstore 
    CMake Error at utilities/bladeRF-cli/CMakeLists.txt:104 (message):
      generate.bash failed! Try disabling BUILD_BLADERF_CLI_INTERACTIVE_HELP, or
      fix the above error.
    -- Configuring incomplete, errors occurred!
     See also "D:/src/sdrangel/build/external/bladerf/src/bladerf-build/CMakeFiles/CMakeOutput.log".
     See also "D:/src/sdrangel/build/external/bladerf/src/bladerf-build/CMakeFiles/CMakeError.log".

If you don't have a Blade RF SDR, you can disable support for it (and hopefully skip this error) with:

    cmake -DENABLE_BLADERF=OFF 

You might also want to add:

    -DENABLE_HACKRF=OFF -DENABLE_IIO=OFF -DENABLE_MIRISDR=OFF -DENABLE_PERSEUS=OFF -DENABLE_XTRX=OFF

arfy8820, 2023-04-29:
Yup, taking out all the extra SDR support got things to move on to.

 

[132/2334] Linking CXX shared library bin\sdrbase.dll                                                                                                                                                                                        

FAILED: bin/sdrbase.dll lib/sdrbase.lib                                                                                                                                                                                                      

cmd.exe /C "cd . && D:\Qt\Tools\CMake_64\bin\cmake.exe -E vs_link_dll --intdir=sdrbase\CMakeFiles\sdrbase.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100226~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100226~1.0\x64\mt.exe --manifests  -- C:\PROGRA~2\MICROS~2\2022\BUILDT~1\VC\Tools\MSVC\1435~1.322\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\sdrbase.rsp  /out:bin\sdrbase.dll /implib:lib\sdrbase.lib /pdb:bin\sdrbase.pdb /dll /version:0.0 /INCREMENTAL:NO /LTCG  && cd ."     

LINK: command "C:\PROGRA~2\MICROS~2\2022\BUILDT~1\VC\Tools\MSVC\1435~1.322\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\sdrbase.rsp /out:bin\sdrbase.dll /implib:lib\sdrbase.lib /pdb:bin\sdrbase.pdb /dll /version:0.0 /INCREMENTAL:NO /LTCG /MANIFEST /MANIFESTFILE:bin\sdrbase.dll.manifest" failed (exit code 1120) with the following output:                                                                                                                                        

   Creating library lib\sdrbase.lib and object lib\sdrbase.exp                                                                                                                                                                               

LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library                                                                                                                                      

webapirequestmapper.cpp.obj : error LNK2001: unresolved external symbol "void __cdecl boost::throw_exception(class std::exception const &)" ***@***.***@@***@***.***@@@Z)                                               

bin\sdrbase.dll : fatal error LNK1120: 1 unresolved externals                                                                                                                                                                                

ninja: build stopped: subcommand failed.                                                                                                                                                                                                     

 

 

So this time it looks like the linker isn’t getting references to boost, even though the boost path was found in the cmake step. This was using the prebuilt boost binaries for visual studio that are available. Should I redo things and boost my own copy? I did try this, but it seems to insist on wanting to build a 32-bit boost.

Yeah, I’m a little bit noob when dealing with windows dev, I’m more used to a unix environment.

 

https://aka.ms/wslstore

srcejon, 2023-05-01:
Where did you download boost from?

I just built sucessfully with: https://boostorg.jfrog.io/artifactory/main/release/1.82.0/source/boost_1_82_0.zip

arfy8820, 2023-05-01:
strange! I believe I have the exact same file. Is the error that I haven't run bootstrap and built boost? Or are you not using any of the parts of boost that require libraries. Also. My x64 native command prompt says its visual studio tools 17.5.4.

arfy8820, 2023-05-01:
aha! Seems like removing directories (boost directory and the build directory) and recompiling did the trick. We're in action now.

closed 2023-05-01 by arfy8820