f4exb/sdrangel #1682: 10kHz doppler errors

Valstee, 2023-05-02:
Hi @f4exb and @srcejon , 
I use sdrangel (SDRangel  7.8.3-8-g0c9686511  Qt 5.15.3  x86_64  Ubuntu 22.04.2 LTS) to receive amateur radio satellites like uvsqsat, inspiresat-7 and many others.
I don't know how to explain it but lately I have been getting regular 10kHz doppler errors when uvsqsat and inspiresat-7 are passing by simultaneously.
The prioritization works well, i.e. it targets the right satellite but the doppler starts at +/- 20kHz and goes to about 0Hz which is fine but 10kHz off... 
I have made presets for each of the satellites and when they don't arrive at the same time I don't have this problem. 
Do you know where this can come from?
Thanks,
F4JIZ

Valstee, 2023-05-17:
yes sorry, I waited to have again the same configuration (i.e. a simultaneous passage of the 2 satellites).

```
2023-05-17 11:24:43.755 (D) SatelliteTrackerWorker::doppler  "UVSQ-SAT"
2023-05-17 11:24:43.756 (D) SSBDemod::webapiSettingsPutPatch: forward to GUI: 0x55c975d36638
2023-05-17 11:24:43.756 (D) SatelliteTrackerWorker::doppler  "INSPIRE-SAT 7"
2023-05-17 11:24:43.756 (D) SSBDemod::webapiSettingsPutPatch: forward to GUI: 0x55c975d36638
2023-05-17 11:24:43.760 (D) SSBDemod::handleMessage: MsgConfigureSSBDemod
2023-05-17 11:24:43.761 (D) SSBDemod::applySettings:  m_inputFrequencyOffset:  -938  m_filterIndex:  0  [m_spanLog2:  1  m_rfBandwidth:  18000  m_lowCutoff:  300  m_fftWindow:  7 ]  m_volume:  1  m_audioBinaual:  false  m_audioFlipChannels:  false  m_dsb:  false  m_audioMute:  false  m_agcActive:  false  m_agcClamping:  false  m_agcTimeLog2:  7  agcPowerThreshold:  -100  agcThresholdGate:  4  m_audioDeviceName:  "System default device"  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
2023-05-17 11:24:43.761 (D) SSBDemod::handleMessage: MsgConfigureSSBDemod
2023-05-17 11:24:43.761 (D) SSBDemodBaseband::handleMessage: MsgConfigureSSBDemodBaseband
2023-05-17 11:24:43.761 (D) DownChannelizer::applyChannelization done:  nb stages: 0  in (baseband): 64000  req: 48000  out (channel): 64000  fc: -938
2023-05-17 11:24:43.761 (D) SSBDemodSink::applyChannelSettings:  channelSampleRate:  64000  channelFrequencyOffset:  -938
2023-05-17 11:24:43.761 (D) NCOF::setFreq: freq: 938.000000 sr: 64000.000000 m_phaseIncrement: 60.032001
2023-05-17 11:24:43.761 (D) SSBDemodSink::applySettings:  m_inputFrequencyOffset:  -938  m_filterIndex:  0  [m_spanLog2:  1  m_rfBandwidth:  18000  m_lowCutoff:  300  m_fftWindow:  7 ]  m_volume:  1  m_audioBinaual:  false  m_audioFlipChannels:  false  m_dsb:  false  m_audioMute:  false  m_agcActive:  false  m_agcClamping:  false  m_agcTimeLog2:  7  agcPowerThreshold:  -100  agcThresholdGate:  4  m_audioDeviceName:  "System default device"  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
2023-05-17 11:24:43.761 (D) SSBDemod::applySettings:  m_inputFrequencyOffset:  -10329  m_filterIndex:  0  [m_spanLog2:  1  m_rfBandwidth:  18000  m_lowCutoff:  300  m_fftWindow:  7 ]  m_volume:  1  m_audioBinaual:  false  m_audioFlipChannels:  false  m_dsb:  false  m_audioMute:  false  m_agcActive:  false  m_agcClamping:  false  m_agcTimeLog2:  7  agcPowerThreshold:  -100  agcThresholdGate:  4  m_audioDeviceName:  "System default device"  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false

```

srcejon, 2023-05-23:
Oops - that patch has a bug. Will fix in a bit.

Valstee, 2023-06-05:
Hello @srcejon ,
Thank you very much for the patch! 🥇 
Only now the problem is that the doppler correction doesn't trigger every time... I'll get the logs.

Valstee, 2023-06-07:
```
2023-06-05 12:23:45.684 (D) SatelliteTrackerWorker::doppler  "UVSQ-SAT"
2023-06-05 12:23:45.684 (D) SSBDemod::webapiSettingsPutPatch: forward to GUI: 0x561cb4f50950
2023-06-05 12:23:45.684 (D) SatelliteTrackerWorker::doppler  "UVSQ-SAT"
2023-06-05 12:23:45.685 (D) SSBDemod::webapiSettingsPutPatch: forward to GUI: 0x561cb4f50950
2023-06-05 12:23:45.690 (D) GS232Controller::handleMessage: MsgConfigureGS232Controller
2023-06-05 12:23:45.690 (D) GS232Controller::applySettings: " m_azimuth: 340 m_elevation: 6 m_reverseAPIAddress: 127.0.0.1"  force:  false
2023-06-05 12:23:45.690 (D) GS232ControllerWorker::applySettings: " m_azimuth: 340 m_elevation: 6 m_reverseAPIAddress: 127.0.0.1"  force:  false
2023-06-05 12:23:45.690 (D) SSBDemod::handleMessage: MsgConfigureSSBDemod
2023-06-05 12:23:45.690 (D) SSBDemod::applySettings:  m_inputFrequencyOffset:  -21  m_filterIndex:  0  [m_spanLog2:  1  m_rfBandwidth:  18000  m_lowCutoff:  300  m_fftWindow:  7 ]  m_volume:  1  m_audioBinaual:  false  m_audioFlipChannels:  false  m_dsb:  false  m_audioMute:  false  m_agcActive:  false  m_agcClamping:  false  m_agcTimeLog2:  7  agcPowerThreshold:  -100  agcThresholdGate:  4  m_audioDeviceName:  "System default device"  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
2023-06-05 12:23:45.690 (D) SSBDemodBaseband::handleMessage: MsgConfigureSSBDemodBaseband
2023-06-05 12:23:45.690 (D) DownChannelizer::applyChannelization done:  nb stages: 0  in (baseband): 64000  req: 48000  out (channel): 64000  fc: -21
2023-06-05 12:23:45.690 (D) SSBDemodSink::applyChannelSettings:  channelSampleRate:  64000  channelFrequencyOffset:  -21
2023-06-05 12:23:45.690 (D) NCOF::setFreq: freq: 21.000000 sr: 64000.000000 m_phaseIncrement: 1.344000
2023-06-05 12:23:45.691 (D) SSBDemodSink::applySettings:  m_inputFrequencyOffset:  -21  m_filterIndex:  0  [m_spanLog2:  1  m_rfBandwidth:  18000  m_lowCutoff:  300  m_fftWindow:  7 ]  m_volume:  1  m_audioBinaual:  false  m_audioFlipChannels:  false  m_dsb:  false  m_audioMute:  false  m_agcActive:  false  m_agcClamping:  false  m_agcTimeLog2:  7  agcPowerThreshold:  -100  agcThresholdGate:  4  m_audioDeviceName:  "System default device"  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
```

srcejon, 2023-06-07:
Can you describe the problem a little more - that log appears to be showing doppler being applied to SSBDemod. E.g.  m_inputFrequencyOffset: is set to -21  (Although looks like it might be doing it twice unecessarily)

For additional debug, you could add the following line:

    qDebug() << "SatelliteTrackerWorker::doppler " << satWorkerState->m_name << "initOffset" << initOffset << "doppler" << doppler << "offset" << offset;

In plugins/feature/satellitetracker/satellitetrackerworker.cpp on line 922 (Just before if (!ChannelWebAPIUtils::setFrequencyOffset ...)

Valstee, 2023-06-26:
Sorry for my late reply, but for some reason the bug doesn't show up every time.
 I'll add the line and let you know. thanks