f4exb/sdrangel #1689: segfault with audiooutput plugin

dforsi, 2023-05-06:
Steps to reproduce:

- open a new workspace (menu: Workspaces / New)
- add an audio output (button: Add Tx device)
- enable "reverse API" (button: cog) 
- click Ok
- segfault

Backtrace:
```
Thread 1 "sdrangel" received signal SIGSEGV, Segmentation fault.
0x00007ffff569bdba in QNetworkAccessManager::sendCustomRequest(QNetworkRequest const&, QByteArray const&, QIODevice*) () from /lib/x86_64-linux-gnu/libQt5Network.so.5
(gdb) bt
#0  0x00007ffff569bdba in QNetworkAccessManager::sendCustomRequest(QNetworkRequest const&, QByteArray const&, QIODevice*) () from /lib/x86_64-linux-gnu/libQt5Network.so.5
#1  0x00007fff982865fb in AudioOutput::webapiReverseSendSettings(QList<QString> const&, AudioOutputSettings const&, bool) ()
   from /home/test/sdrangel/build/lib/plugins/liboutputaudio.so
#2  0x00007fff98287222 in AudioOutput::applySettings(AudioOutputSettings const&, QList<QString> const&, bool) () from /home/test/sdrangel/build/lib/plugins/liboutputaudio.so
#3  0x00007fff982880a4 in AudioOutput::handleMessage(Message const&) () from /home/test/sdrangel/build/lib/plugins/liboutputaudio.so
#4  0x00007ffff7364650 in DeviceSampleSink::handleInputMessages() () from /home/test/sdrangel/build/lib/libsdrbase.so
#5  0x00007ffff5ee8f7c in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007fff9828ac19 in AudioOutputGui::updateHardware() () from /home/test/sdrangel/build/lib/plugins/liboutputaudio.so
#7  0x00007ffff5ee8f7c in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff5eecd6a in QTimer::timeout(QTimer::QPrivateSignal) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff5edd50d in QObject::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff6b62fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff5eb16f8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff5f08c31 in QTimerInfoList::activateTimers() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff5f094c4 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff451e7a9 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff451ea38 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007ffff451eacc in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff5f09836 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff5eb017b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff5eb82d6 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x0000555555556ee8 in runQtApplication(int, char**, qtwebapp::LoggerWithFile*) ()
#21 0x0000555555556491 in main ()
```

srcejon, 2023-05-16:
Looks to be the same problem as was here: https://github.com/f4exb/sdrangel/issues/1144

m_networkManager doesn't appear to be initialised.

f4exb, 2023-05-18:
yes... I did a systematic search and found other occurrences. Should be an easy fix.

commit 1247e7e6 (2023-05-18): Fixed missing initializations of QNetworkManager. Fixes #1689
closed 2023-05-18 by f4exb