f4exb/sdrangel #1690: Pluto Tx infinite loop

drom, 2023-05-08:
Installed latest: `SDRangel 7.13.0 Qt 5.15.9 64b x86_64 openSUSE Tumbleweed`
I can add Pluto RX channel fine. But after adding Pluto TX, SDRangel goes into infinite loop with this message repeated again and again:
```
...
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999956: 9
DevicePlutoSDRBox::set_params: set attribute out_altvoltage0_RX_LO_frequency=435000000: 10
...
```

eu2aa, 2023-05-08:
Accidentally updated SDRangel to version 7.13.0-1.8 and got a problem. TRANSMISSION window won't load, persistent message, here's a snippet: 

2023-05-08 21:26:45.667 (I) PluginManager::loadPluginsDir: loaded plugin libudpsource.so
2023-05-08 21:26:45.668 (I) PluginManager::loadPluginsDir: loaded plugin libvorlocalizer.so
[INFO] [UHD] linux; GNU C++ version 13.0.1 20230421 (prerelease) [revision f980561c60b0446cc427595198d7f3f4f90e0924]; Boost_108200; UHD_4.4.0.0-0-unknown
21:26:47.264855 WARN:   xtrxllpciev0_discovery:243 [PCIE] XTRX PCIe driver isn't loaded
21:26:48.808536 WARN:   xtrxllpciev0_discovery:243 [PCIE] XTRX PCIe driver isn't loaded
21:26:50.336834 WARN:   xtrxllpciev0_discovery:243 [PCIE] XTRX PCIe driver isn't loaded
2023-05-08 21:26:50.355 (I) WebAPIServer::start: starting web API server at http://:8091
2023-05-08 21:26:50.356 (W) qUncompress: Input data is corrupted
2023-05-08 21:26:50.356 (W) qUncompress: Input data is corrupted
==>
==>
==>
2023-05-08 21:26:53.942 (W) libpng warning: iCCP: known incorrect sRGB profile
2023-05-08 21:26:53.942 (W) libpng warning: iCCP: too many profiles
==>
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999956: 9
DevicePlutoSDRBox::set_params: set attribute out_altvoltage0_RX_LO_frequency=435000000: 10
DevicePlutoSDRBox::set_params: set attribute in_voltage_rf_bandwidth=1500000: 8
DevicePlutoSDRBox::set_params: set attribute in_voltage0_rf_port_select=A_BALANCED: 11
DevicePlutoSDRBox::set_params: set attribute in_voltage0_gain_control_mode=manual: 7
DevicePlutoSDRBox::set_params: set attribute in_voltage0_hardwaregain=40: 3
DevicePlutoSDRBox::set_params: set attribute in_voltage_bb_dc_offset_tracking_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_rf_dc_offset_tracking_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_quadrature_tracking_en=1: 2
==>
==>
2023-05-08 21:26:55.813 (W) AudioOutputDevice::start: using system default device alsa_output.pci-0000_00_1f.3.hdmi-stereo
2023-05-08 21:26:55.813 (I) AudioOutputDevice::start: audio format OK
2023-05-08 21:26:55.882 (W) libpng warning: iCCP: known incorrect sRGB profile
2023-05-08 21:26:55.882 (W) libpng warning: iCCP: too many profiles
==>
==>
2023-05-08 21:26:55.952 (W) libpng warning: iCCP: known incorrect sRGB profile
2023-05-08 21:26:55.952 (W) libpng warning: iCCP: too many profiles
==>
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999956: 9
DevicePlutoSDRBox::set_params: set attribute out_altvoltage1_TX_LO_frequency=435000000: 10
DevicePlutoSDRBox::set_params: set attribute out_voltage_rf_bandwidth=1500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage0_rf_port_select=A: 2
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-12.5: 6
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999956: 9
DevicePlutoSDRBox::set_params: set attribute out_altvoltage0_RX_LO_frequency=435000000: 10
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
[... 31 lines cut ...]
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999956: 9
DevicePlutoSDRBox::set_params: set attribute out_altvoltage0_RX_LO_frequency=435000000: 10
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
[... cut ...]
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999956: 9
DevicePlutoSDRBox::set_params: set attribute out_altvoltage0_RX_LO_frequency=435000000: 10
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox

How to return to the old version?

eu2aa, 2023-05-16:
I had to switch to the old version 7.6.3-lp153.1.3.x86_64

eu2aa, 2023-05-16:
Version 7.13.0-1.9 has the same problem

srcejon, 2023-05-16:
> I had to switch to the old version 7.6.3-lp153.1.3.x86_64

That suggests the problem was introduced in this patch:

https://github.com/f4exb/sdrangel/commit/94a465b2108639ad70f685fd0c6afcb6642d0283

Possibly one of the settings keys is wrong / missing.

(I don't have a Pluto so can't debug it.)

eu2aa, 2023-05-16:
interesting idea

mattmelling, 2023-05-16:
> Possibly one of the settings keys is wrong / missing.

Is this settings as in user configuration? Can it be cleared by removing some config file?

> That suggests the problem was introduced in this patch:
>
> https://github.com/f4exb/sdrangel/commit/94a465b2108639ad70f685fd0c6afcb6642d0283

Can also confirm that 7.6.0 works. while 7.9.0 does not which also points to that commit.

srcejon, 2023-05-16:
> > Possibly one of the settings keys is wrong / missing.
> 
> Is this settings as in user configuration? Can it be cleared by removing some config file?

No. I suspect it's a problem in the code.

drom, 2023-06-26:
Anybody having similar issue?

eu2aa, 2023-06-26:
too bad there won't be a FIX for TUMBLEWEED.
I'm still working with the surrogate version 7.6.3-lp153.1.3.x86_64.
Transceiver mode has fallen.
I'll have to go somewhere...
it's a pity,
that did not wait for the solution of the problem.
Good luck everyone!

srcejon, 2023-07-03:
One possible problem in plutosdroutput.cpp. We have:

    else if (DevicePlutoSDRShared::MsgCrossReportToBuddy::match(message)) // message from buddy
    {
        DevicePlutoSDRShared::MsgCrossReportToBuddy& conf = (DevicePlutoSDRShared::MsgCrossReportToBuddy&) message;
        m_settings.m_devSampleRate = conf.getDevSampleRate();
        m_settings.m_lpfFIRlog2Interp = conf.getLpfFiRlog2IntDec();
        m_settings.m_lpfFIRBW = conf.getLpfFirbw();
        m_settings.m_LOppmTenths = conf.getLoPPMTenths();
        PlutoSDROutputSettings newSettings = m_settings;
        newSettings.m_lpfFIREnable = conf.isLpfFirEnable();

        applySettings(newSettings, QList<QString>{"lpfFIREnable"});

        return true;
    }

This call to applySettings appears to be missing some settings keys. Just like in the input module, it should probably be:

        applySettings(newSettings, QList<QString>{"devSampleRate", "lpfFIRlog2Decim", "lpfFIRBW", "LOppmTenths", "lpfFIREnable"});

f4exb, 2023-07-11:
The loop appears because the input wants to send some changes to the output and then the output to the input etc...
```
2023-07-11 10:33:23.195 (D) PlutoSDRInput::applySettings: forwardChangeOtherDSP
2023-07-11 10:33:23.195 (D) PlutoSDROutput::applySettings: force: false  m_LOppmTenths: 0 m_devSampleRate: 2500000 m_lpfFIREnable: 0 m_lpfFIRBW: 500000 m_lpfFIRlog2Interp: 0
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
2023-07-11 10:33:23.307 (D) DevicePlutoSDRBox::getRxSampleRates: BBPLL:959999999 ADC:29999999 R2:9999999 R1:4999999 RF:2499999 RXSAMP:2499999
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
2023-07-11 10:33:23.335 (D) DevicePlutoSDRBox::setFIR: intdec: 1 gain: 0 nbTaps: 64 BWin: 500000 BW: 499999.993715 (0.200000)
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
2023-07-11 10:33:23.411 (D) PlutoSDRInput::applySettings: BBPLL(Hz):  959999999  DAC:  29999999  <-HB3-  9999999  <-HB2-  4999999  <-HB1-  2499999  <-FIR-  2499999
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999913: 9
2023-07-11 10:33:23.411 (D) PlutoSDROutput::applySettings: forwardChangeOtherDSP
2023-07-11 10:33:23.411 (D) PlutoSDRInput::applySettings: force:  false " m_LOppmTenths: 0 m_devSampleRate: 2500000 m_lpfFIREnable: 0 m_lpfFIRBW: 500000 m_lpfFIRlog2Decim: 0"
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
2023-07-11 10:33:23.496 (D) DevicePlutoSDRBox::getRxSampleRates: BBPLL:959999999 ADC:29999999 R2:9999999 R1:4999999 RF:2499999 RXSAMP:2499999
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
2023-07-11 10:33:23.522 (D) DevicePlutoSDRBox::setFIR: intdec: 1 gain: 0 nbTaps: 64 BWin: 500000 BW: 499999.993715 (0.200000)
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
2023-07-11 10:33:23.602 (D) DevicePlutoSDRBox::getRxSampleRates: BBPLL:959999999 ADC:29999999 R2:9999999 R1:4999999 RF:2499999 RXSAMP:2499999
2023-07-11 10:33:23.602 (D) PlutoSDRInput::applySettings: BBPLL(Hz):  959999999  ADC:  29999999  -HB3->  9999999  -HB2->  4999999  -HB1->  2499999  -FIR->  2499999
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999913: 9
2023-07-11 10:33:23.602 (D) DeviceSampleStatic::calculateSourceDeviceCenterFrequency:  frequencyShiftScheme:  0  desired center freq:  435000000  Hz  device center freq:  435000000  Hz  device sample rate:  2500000 S/s  Actual sample rate:  2500000 S/s  center freq position code:  2  image frequency:  435000000 Hz
DevicePlutoSDRBox::set_params: set attribute out_altvoltage0_RX_LO_frequency=435000000: 10
2023-07-11 10:33:23.604 (D) PlutoSDRInput::applySettings: forwardChangeOtherDSP
```
This design prior to the implementation of "MIMO" devices is problematic because it tries to fit two plugins for the same physical device.

closed 2023-07-11 by f4exb

eu2aa, 2023-07-11:
only  7.13.0 on  https://software.opensuse.org/package/sdrangel, sorry

eu2aa, 2023-07-11:
thank, Edouard!

commit c4debeef (2023-07-11): PlutoSDR input/output: do not apply settings to self when coming from a buddy change. Fixes #1690