f4exb/sdrangel #1691: cmake list misses a necessary package

csylvain, 2023-05-10:
{Ubuntu 22.04}
build stops because of missing header file
```
[homedir]/Software/sdrangel/sdrbase/util/osndb.cpp:21:10: fatal error: QtGui/private/qzipreader_p.h: No such file or directory
   21 | #include <QtGui/private/qzipreader_p.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/build.make:2155: sdrbase/CMakeFiles/sdrbase.dir/util/osndb.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:4205: sdrbase/CMakeFiles/sdrbase.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```
cmake list should check for package qtbase5-private-dev
```
$ apt-file find qzipreader_p.h
qtbase5-private-dev: /usr/include/x86_64-linux-gnu/qt5/QtGui/5.15.3/QtGui/private/qzipreader_p.h
```

f4exb, 2023-05-15:
https://packages.debian.org/sid/qtbase5-private-dev

"This package contains the private header development files for building some Qt 5 applications like Qt Creator QML Designer plugin."

So this should be a dependency only for the Qt ecosystem applications. Please check your Qt installation.

f4exb, 2023-05-15:
I think this is not possible to detect, This threads concludes that the problem persists after various workarounds were attempted: https://github.com/LMMS/lmms/issues/4273

I suggest you follow instructions in the wiki https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux where this dependency is explicitly stated.

closed 2023-05-15 by f4exb

csylvain, 2023-05-16:
> problem persists after various workarounds were attempted: LMMS/lmms#4273

"Please check your Qt installation."  <-- it was as Ubuntu 22.04 installs
packages to run Qt Apps.

I made changes as Cmake indicated after cloning the repo (added packages
which were not found by Cmake).

https://github.com/LMMS/lmms/issues/4273#issuecomment-427247129 appears to
show a workaround equivalent to simple hack of checking
in Cmake the file QtGui/private/qzipreader_p.h exists. It is bad for Cmake
to generate a Makefile when there are missing necessary components, yes?

oh, and yes, thank you for pointing out the explicit list of dependencies
in the Wiki.
i suppose it may have been better to have cut-and-pasted rather than let
Cmake guide me to missing packages.  :-/