f4exb/sdrangel #1692: Crash on refreshing available devices

tko, 2023-05-11:
I built 7.13.0 on M1 mac with sdrplay_api3 and for most parts everything seems to be going fine. However, at some point my SDRPlayV3 disappeared from list of available devices, and now every time I try to refresh the listing I get a crash.

```
2023-05-11 19:18:04.363 sdrangel[37309:2533984] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00000001844ed138 __exceptionPreprocess + 240
	1   libobjc.A.dylib                     0x0000000184237e04 objc_exception_throw + 60
	2   CoreFoundation                      0x00000001845180bc _CFBundleGetValueForInfoKey + 0
	3   AppKit                              0x0000000186ffb8cc -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 372
	4   AppKit                              0x000000018700a6e8 -[NSThemeFrame _tileTitlebarAndRedisplay:] + 132
	5   AppKit                              0x000000018702680c -[NSThemeFrame setFrameSize:] + 416
	6   AppKit                              0x0000000187025eec -[NSWindow _oldPlaceWindow:fromServer:] + 568
	7   AppKit                              0x0000000187024d7c -[NSWindow _setFrameCommon:display:fromServer:] + 2220
	8   libqcocoa.dylib                     0x000000010120c6b4 qt_plugin_instance + 72012
	9   libqcocoa.dylib                     0x000000010120c450 qt_plugin_instance + 71400
	10  QtGui                               0x00000001027afcbc _ZN7QWindow6resizeERK5QSize + 1124
	11  QtGui                               0x00000001027ae430 _ZN7QWindow6resizeEii + 24
	12  QtWidgets                           0x00000001022242f8 _ZN14QWidgetPrivate15setGeometry_sysEiiiib + 1172
	13  QtWidgets                           0x000000010222461c _ZN7QWidget6resizeERK5QSize + 120
	14  QtWidgets                           0x00000001024130c4 _ZN15QProgressDialog12setLabelTextERK7QString + 132
	15  libsdrgui.dylib                     0x00000001007a8aa8 _ZN26SamplingDeviceDialogWorker18enumeratingDevicesERK7QString + 68
	16  QtCore                              0x0000000102f1a2fc _ZN11QMetaObject8activateEP7QObjectPKS_iPPv + 3204
	17  libsdrbase.dylib                    0x0000000101303c28 _ZN16DeviceEnumerator18enumeratingDevicesERK7QString + 52
	18  libsdrbase.dylib                    0x00000001013656c4 _ZN16DeviceEnumerator16enumerateDevicesER5QListIN9PluginAPI26SamplingDeviceRegistrationEERNSt3__16vectorINS_17DeviceEnumerationENS5_9allocatorIS7_EEEEN15PluginInterface14SamplingDevice10StreamTypeE + 144
	19  libsdrgui.dylib                     0x00000001007a89e8 _ZN26SamplingDeviceDialogWorker16enumerateDevicesEv + 184
	20  QtCore                              0x0000000102f1a2fc _ZN11QMetaObject8activateEP7QObjectPKS_iPPv + 3204
	21  QtCore                              0x0000000102d26fe4 _ZN7QThread7startedENS_14QPrivateSignalE + 56
	22  QtCore                              0x0000000102d27888 _ZN7QThread11qt_metacallEN11QMetaObject4CallEiPPv + 1276
	23  libsystem_pthread.dylib             0x00000001843a026c _pthread_start + 148
	24  libsystem_pthread.dylib             0x000000018439b08c thread_start + 8
)
libc++abi: terminating with uncaught exception of type NSException
```

tko, 2023-05-11:
Commenting out the line in stack trace makes the crash go away so presumably that callback should signal main thread instead to do the update.

(For some reason I'm not seeing the device listed even though it looks ok in plugins list and is functional in a different workspace.)

```diff
diff --git a/sdrgui/gui/samplingdevicedialog.cpp b/sdrgui/gui/samplingdevicedialog.cpp
index 5b2aeb6..cf51849 100644
--- a/sdrgui/gui/samplingdevicedialog.cpp
+++ b/sdrgui/gui/samplingdevicedialog.cpp
@@ -133,6 +133,6 @@ void SamplingDeviceDialogWorker::enumerateDevices()
 
 void SamplingDeviceDialogWorker::enumeratingDevices(const QString &deviceId)
 {
-    m_progressDialog->setLabelText("Enumerating " + deviceId);
+    // m_progressDialog->setLabelText("Enumerating " + deviceId);
 }
 
```

srcejon, 2023-05-11:
> (For some reason I'm not seeing the device listed even though it looks ok in plugins list and is functional in a different workspace.)

You can't have the same device in two workspaces.

MartyF81, 2023-05-24:
I am facing the same issue.  Is there a version of this I can roll back to before this change was implemented?

srcejon, 2023-05-24:
> Commenting out the line in stack trace makes the crash go away so presumably that callback should signal main thread instead to do the update.

I don't see the crash on Windows or Linux, but agree, the UI update should be on the UI thread, so the above patch should change that.

PR #1697 (2023-05-24): Sampling Device Dialog. Only update widgets from UI thread.
In Sampling Device Dialog, only update widgets from UI thread. May fix #1692.
closed 2023-05-24 by f4exb