f4exb/sdrangel #1699: SigMFFileInput acceleration doesn't appear to work

srcejon, 2023-05-25:
The acceleration factor in the SigMFFileInput doesn't appear to work. Regardless of the setting (E.g. 1 or 10), playback appears to always be in real-time.

srcejon, 2023-05-25:
It also seems that the "Full Time navigator" doesn't work. If I adjust the time using the slider, the waveform appears to continue from the start of the file, rather than the set time position. 

If I use the "Track Time Navigator" slider instead (the one on top), it appears to work correctly.

f4exb, 2023-06-03:
Above commit fixes the acceleration problem. I haven't seen any issue with the "Full Time navigator" as for me it navigates through the whole record as expected.

srcejon, 2023-06-03:
> Above commit fixes the acceleration problem. I haven't seen any issue with the "Full Time navigator" as for me it navigates through the whole record as expected.

Thanks.

For a (large) test case, you should be able to download A_Sign_in_Space-ATA-X.sigmf-data from https://bldata.berkeley.edu/A_Sign_in_Space/

This file contains a signal that is doppler shifted over the length of the file (moves from high to low frequency). If you set a marker at the signal's frequency at the start of the file, then use the two navigator scroll bars to jump around the file, you should hopefully see that whenever using the Full Time Navigator, the signal returns to the marker's frequency (i.e. the start of the file).

f4exb, 2023-06-05:
I see... the slider moves accordingly but in fact it's a lie. However on a multi track recording the track corresponding to the position in the whole file is selected/highlighted properly. It seems that in this case the file seek does not work as expected.