f4exb/sdrangel #1701: SDR Angel v 7.14.1 failed to start on windows 7

morphevsfx, 2023-05-26:
Hello,

I installed the latest SDR Anger ver on windows 7 and it fails to start.
Installed with admin rights and start the application with admin rights

Faulting application name: sdrangel.exe, version: 0.0.0.0, time stamp: 0x646db76d
Faulting module name: qtposition_winrt.dll, version: 5.15.2.0, time stamp: 0x5faa701f
Exception code: 0xc0000005
Fault offset: 0x00000000000019a2
Faulting process id: 0x22a8
Faulting application start time: 0x01d98fc5473c6d38
Faulting application path: C:\Program Files\SDRangel\sdrangel.exe
Faulting module path: C:\Program Files\SDRangel\position\qtposition_winrt.dll
Report Id: 852c0428-fbb8-11ed-b3f9-d5e06a39ac95

srcejon, 2023-06-06:
From the cmd.exe, try:

    cd "c:\Program Files\SDRangel"
    .\sdrangel.exe --scratch

morphevsfx, 2023-07-18:
Nope didn't work.
The process starts, no windows showing. And after a while process just dies

ᐧ

N5BRG, 2023-07-19:
Morphevs,

Take a look in /var/logs/syslog and see if you can get a idea what is causing the problem. The listing related will be at the bottom of the file just after you run the code.

You probably have some issue with a library that changed or was moved in an update and it only on your system. It would not be possible for someone to find this for you.
You could also try rebuild the code on your computer and to correct your issues.

good luck with it.

Bob

closed 2023-09-04 by github-actions[bot]