f4exb/sdrangel #1702: Frequencies above 48 kHz with Audio Input plugin?

svenfl, 2023-05-26:
Is it possible to provide sample frequencies above 48 kHz with the existing or a new audio plugin to provide frequencies up to 384 kHz like audacity does. The backround of my question is trying to decode DCF77 radio clock at 77.5 kHz. For succesful reaching the target I need at least a sample rate of 145 kHz or above. 192 kHz would be useful to proceed. I connect just a short wire to the microphone jack of the soundcard and can create a sample using audacity but that is not my target because it is not real time decoding. I am not a programmer so I am not able to write the plugin myself. Perhaps that new or modified plugin would be useful to recive other radio clock at 60 kHz or 162 kHz as well.

srcejon, 2023-05-26:
It already does for me on Windows.

What OS are you using and what audio device do you have selected? A screenshot might help.

svenfl, 2023-05-26:
I use 3 different OS Linux, MAC OS 10.15.7 and Windows 10
On Windows 10 indeed it is possible to coose SR 192000 Hz for input. 
But there is no possibility to set a frequency or a frequency range.
Wanted frequency is 77.5 kHz or the range from 64 to 96 kHz in spectrum.
Device is a Realtek High Definition Audio on Windows 10
On Ubuntu on the same Computer I can only choose sample rate up to 48kHz and no possibility to specify a frequency to listen on. 
MAC OS is on my MacBook with the same Limitation of SR up to 48 kHz

srcejon, 2023-05-26:
If I recall correctly,  the sample rates are just what Qt lists are available,  but I should check.

But if your wanted signal is at 77.5kHz, then just use 192kHz. Then put the radio clock demod at an offset of 77.5kHz

f4exb, 2023-05-27:
In the "Audio Preferences" menu you can set the sample rate of the soundcard:
[image]

This will work or fail depending on what Pulseaudio (that sits behind Qt) can do but if your soundcard is natively 192 or 384 kHz this should work. For example I can do the opposite and set my native 48 kHz soundcard to 8 kHz.

svenfl, 2023-05-29:
I will have look on pulseaudio. Perhaps my soundcard is not correctly configured or does not suppotz higher sample rates above 48 kHz on linux or mac os. Thanks for your answers.

RobRich999, 2023-05-30:
The Arch Linux wiki has good info about Linux audio configuration that works for many mainstream distros.

If actually running PulseAudio, you probably need to edit /etc/pulse/daemon.conf (or a copy in ~/.config/pulse/ for user level) to change default-sample-rate, default-sample-format, and/or resample-method.

https://wiki.archlinux.org/title/PulseAudio

Alternatively, if using[pipewire-pulse support in PipeWire, you probably will need to copy the various PipeWire config files to ~/.config (user level) or /etc/pipewire (system level), the modify the allowed sample rates, resampling method, etc.

https://wiki.archlinux.org/title/PipeWire

closed 2023-07-16 by github-actions[bot]