f4exb/sdrangel #1707: DSC decoder plugin FSK frequencies/baud rate incorrect for VHF

tttonyyy, 2023-06-02:
Love the DSC decoder plugin concept - but: 

```
    static const int DSCDEMOD_CHANNEL_SAMPLE_RATE = 1000; // Must be integer multiple of baud rate (x10)
    static const int DSCDEMOD_BAUD_RATE = 100;
    static const int DSCDEMOD_FREQUENCY_SHIFT = 170;
```    

The ITU spec for VHF band (156.525MHz) is for FSK with centre frequency 1700Hz +-400Hz IE 2100Hz and 1300Hz.  The baud rate in VHF band is 1200.

Similarly, the NFM demod settings for VHF DSC (16K0G2B) should be:
RFBW = 16000
FMd = 5000

If manually using the UDP sample sink to capture DSC bursts I have to set the NFM gain to 0.5 otherwise the signals are distorted, not sure how that applies to this plugin, however.  I can provide baseband recordings if that helps development.

Great work,

Tony

srcejon, 2023-06-02:
If you can upload a recording that would be helpful, as am out of range for VHF here. Thanks.

tttonyyy, 2023-06-02:
Here's a baseband recording of a selective individual call.

[routine_call.zip](https://github.com/f4exb/sdrangel/files/11633747/routine_call.zip)

Address: 233333334
Self: 201444555
Called station rx and tx channel 8

srcejon, 2023-06-03:
Thanks, but that .wav file doesn't appear to contain IQ data. It's just a single channel. Can you make the recording in SDRangel using the File Sink plugin?

tttonyyy, 2023-06-03:
No problem. I'm away from home most of today but will get that recording for you as soon as possible.

tttonyyy, 2023-06-03:
Hi @srcejon, try this one:

[selective_individual_call_2.2023-06-03T19_21_59_833.sdriq.zip](https://github.com/f4exb/sdrangel/files/11643536/selective_individual_call_2.2023-06-03T19_21_59_833.sdriq.zip)

sv1, 2024-05-11:
Meanwhile it would be nice to rename the plugin to "HF DSC...." and remove VHF frequency from plugin documentation. This way no frustrated users trying to figure out why their setup does not work!

srcejon, 2026-10-02:
Added support for VHF in this PR: https://github.com/f4exb/sdrangel/pull/2976

But I don't have any other test data other than the above, so let me know if there's still problems.

closed 2026-10-02 by f4exb