f4exb/sdrangel #1709: Can't get 3D models working in the Map

daviderud, 2023-06-06:
Can't make the 3D models display in the 3d map feature in Windows (ver 7.14.2).
I tried performing clean install, removing registry entries and the folders:
C:\Users\USERNAME\AppData\Local\f4exb
C:\Users\USERNAME\AppData\Roaming\f4exb

No luck.

I noticed that the 3d files get extracted in:
C:\data\user\0\org.sdrangel\files\3d\
that seem more like an android path rather than a Windows one...

Thanks

srcejon, 2023-06-06:
Can you post your log file.

srcejon, 2023-06-06:
Oh, and in ADSB settings, check Log 3D model matching information.

daviderud, 2023-06-07:
Ok after deleting the folders in my post above and re-downloading everything with the log enabled, now the 3d files sit in the right folder:
C:\Users\USERNAME\AppData\Roaming\f4exb\SDRangel\3d

And the model works.
I do not know why in the previous attempts the files were extracted into "C:\data\user\0\org.sdrangel\files\3d" instead.

By the way find the logs attached:
- sdrangel.log : after cleaning the folders, run to re-download everything. Interestingly some files failed to download "OpenAIP::downloadFinished: Failed:". Don't know if this is expected.

- sdrangel_postdownload.log: restarted after downloading everything, and started the ADSB demodulation.  Interestingly there are messages "QGeoTileRequestManager: Failed to fetch tile (522,336,10) 5 times, giving up. Last error message was: 'Error transferring https://maps.wikimedia.org/osm-intl/10/522/336.png - server replied: Forbidden: Map tiles are restricted to Wikimedia & affiliated sites only.  See https://wikitech.wikimedia.org/wiki/Maps/External_usage if you believe your usage supports the Movement.'".  In fact I have troubles when zooming the 2D map "OpenStreetMap" 

Anyway with the models files in the AppData\Roaming\ subfolder the 3D models work! 

[sdrangel_logs.zip](https://github.com/f4exb/sdrangel/files/11672505/sdrangel_logs.zip)

srcejon, 2023-06-07:
> * sdrangel.log : after cleaning the folders, run to re-download everything. Interestingly some files failed to download "OpenAIP::downloadFinished: Failed:". Don't know if this is expected.

It's probably OK - there aren't files for all countries.

> Interestingly there are messages "QGeoTileRequestManager: Failed to fetch tile (522,336,10) 5 times, giving up. Last error message was: 'Error transferring [...]

Ok, I can see this to on the Street Map (others seem to still work), which comes from WikiMedia. It seems WikiMedia have restricted the use of the maps to certain sites.

srcejon, 2023-06-07:
The above patch switches tile servers from WikiMedia to OpenStreetMap. Street Map will look a bit different, but should work.

closed 2023-07-24 by github-actions[bot]