f4exb/sdrangel #1714: fatal error C1083: : “hidapi.h”: No such file or directory

dream100200, 2023-06-15:
hi,My project's compilation environment is VS2015 on Windows 10 or Qt 5.5. When I compile the project, it shows that a file(hidapi.h) is missing. Please explain why this is happening. Thank you.

srcejon, 2023-06-15:
Please show the full error message.

Compile instructions are here: https://github.com/f4exb/sdrangel/wiki/Compile-in-Windows - VS2015 isn't recommended.

dream100200, 2023-06-15:
the only error just point out the file of hidapi.h is missed when complied the project with vs2015.
the only err msg: it can not find a file of hidapi.h.

srcejon, 2023-06-15:
If you don't show the full context of the error you are seeing, I can't help you.

dream100200, 2023-06-15:
ok,tks, i check out slowly by Compile instructions are here: https://github.com/f4exb/sdrangel/wiki/Compile-in-Windows - VS2015 isn't recommended.

f4exb, 2023-06-16:
hidapi.h is provided via an external library in the external/windows submodule: https://github.com/f4exb/sdrangel-windows-libraries/tree/a3f4cd7f7e1250a7fa79fce9d5edc573747ca819/hidapi/include 

When cloning the repository did you do a "git submodule init" and "git submodule update" ?

closed 2023-08-02 by github-actions[bot]