f4exb/sdrangel #1715: Plugin: SSH Client

TheBarret, 2023-06-16:
Would it be too ambitious or weird, of having a SSH terminal plugin that can connect to (lan/intra) servers that controls arrays of SDR's?

If it is, then consider me crazy and i'll bugger off :<

f4exb, 2023-06-16:
I just don't really get what you are trying to achieve maybe elaborate a bit more. Having a terminal inside SDRangel (via feature plugin I assume) does not make much sense when you can have one with the O/S.

TheBarret, 2023-06-16:
> I just don't really get what you are trying to achieve maybe elaborate a bit more. Having a terminal inside SDRangel (via feature plugin I assume) does not make much sense when you can have one with the O/S.

Like i said, if it sounds too strange lol

On serious note, your application is modular that i can totally see my self using a build in terminal (Think PuTTy) because I dont know how others setup their devices, I use a dedicated linux server combined with the remote connection, it be nice if all the things are in the same app, but it might be very ambitious.

f4exb, 2023-06-16:
If f this is just a terminal without any connection to the rest of the application I do not see the point. There are many more useful things to develop.

TheBarret, 2023-06-17:
> If f this is just a terminal without any connection to the rest of the application I do not see the point. There are many more useful things to develop.

'without any connection' 🤔 that's a weird assumption
Just consider this never asked you're not the right person indeed 🍦 seeya!

f4exb, 2023-06-17:
> you're not the right person indeed

I take the role of Product Owner of this software so sorry yes I am the right person! It is of no use to make a child fuss if I do not agree on your use case. Instead you should try to prove this has an added value and for now sorry this is not obvious.

For example it does make sense to handle SSH connection in VSCode since you can remotely edit code through SSH. Then you can open a terminal on the same host on where you are editing and it also places you at the root of your source code folder. This is what I mean by "connected to the rest of the application". I still cannot see how this would fit in the SDRangel context.

TheBarret, 2023-06-26:
It was just a question, an idea. It is what it is. good luck mate :) (still love your software tho)

f4exb, 2023-07-01:
then it can be closerd

closed 2023-07-01 by f4exb