f4exb/sdrangel #1722: Question: serial port /dev/ttyUSB0 vs /dev/rfcomm*

choupette-change, 2023-06-27:
Hello, I have a Bluetooth-type GPS, how do I send the GPS-NMEA stream to Sdrangel ? In which file, can we modify the serial port /dev/ttyUSB0 into a serial port of type /dev/rfcomm*?

f4exb, 2023-06-28:
The answer is here: https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/heatmap#readme

"On Windows/Linux/Mac, a GPS supporting NMEA via a serial port at 4800 baud is required. The COM port / serial device should be specified via the QT_NMEA_SERIAL_PORT environment variable before SDRangel is started. (E.g. on Linux: export QT_NMEA_SERIAL_PORT=/dev/ttyACM0 or /dev/ttyUSB0 on Windows: Set QT_NMEA_SERIAL_PORT=COM5 via Control Panel) This requires the Qt serialnmea plugin, which is not available in the libqt5positioning5-plugins package on Debian / Ubuntu. It can be downloaded for Ubuntu/x64 from http://sdrangel.org/downloads/libqtposition_serialnmea.so and should be installed in usr/lib/x86_64-linux-gnu/qt5/plugins/position/. This custom build of the plugin also allows setting the baud rate via the QT_NMEA_SERIAL_BAUD_RATE option (which is not supported on Windows nor Mac)."

f4exb, 2023-06-28:
Maybe the information could be replicated here https://github.com/f4exb/sdrangel/blob/master/sdrgui/readme.md

closed 2023-08-13 by github-actions[bot]