f4exb/sdrangel #1724: SDRangelSrv - Audio issue

Valstee, 2023-06-29:
Hi, 
I use 
```
  "appname": "SDRangelSrv",
  "architecture": "arm64",
  "dspRxBits": 16,
  "dspTxBits": 16,
  "os": "Debian GNU/Linux 11 (bullseye)",
  "pid": 14,
  "qtVersion": "5.15.2",
  "version": "7.15.0-4-ge7cb99843"
```
I'm trying to use sdrangel srv on a raspberry. I've done it several times before, but now I'm getting errors that I can't fix...
I've listed them all below, but I don't know if they're all related...
I think it's similar to several previous issues: https://github.com/f4exb/sdrangel/issues/581 , https://github.com/f4exb/sdrangel/issues/996#issuecomment-913413760 
do you have any idea where this is coming from?
I added these QT libraries during build in the Dockerfile : `qttools5-dev qttools5-dev-tools libspeexdsp-dev libsamplerate0-dev`
it doesn't change anything...

`(W) AudioDeviceManager::setOutputDeviceInfo: index: -1 device: System default device no FIFO registered yet hence no audio output has been allocated yet`

```
(W) AudioOutputDevice::start: using system default device
(W) AudioOutputDevice::start: format 48000 Hz 2xS16LE audio/pcm not supported. Using:  sampleRate: -1 channelCount: -1 sampleSize: -1 codec:  byteOrder: LE sampleType: 0
 (W) AudioOutputDevice::start: Audio device '' failed
```

```
2023-06-29 11:42:31.358 (C) AudioFifo::write: (WFMDemod [0:0]) overflow 12144 samples
2023-06-29 11:42:31.358 (D) WFMDemodSink::feed: 4240/16384 audio samples written
2023-06-29 11:42:31.702 (C) AudioFifo::write: (WFMDemod [0:0]) overflow 16384 samples
2023-06-29 11:42:31.702 (D) WFMDemodSink::feed: 0/16384 audio samples written
2023-06-29 11:42:32.038 (C) AudioFifo::write: (WFMDemod [0:0]) overflow 16384 samples
2023-06-29 11:42:32.039 (D) WFMDemodSink::feed: 0/16384 audio samples written
2023-06-29 11:42:32.382 (C) AudioFifo::write: (WFMDemod [0:0]) overflow 16384 samples
2023-06-29 11:42:32.382 (D) WFMDemodSink::feed: 0/16384 audio samples written
```

`(W) PluginManager::loadPluginsDir: Cannot load library /opt/install/sdrangel/lib/sdrangel/pluginssrv/liblocalsinksrv.so: (/opt/install/sdrangel/lib/sdrangel/pluginssrv/liblocalsinksrv.so: undefined symbol: _ZN17LocalSinkSettings13m_maxFFTBandsE)`

Valstee, 2023-06-29:
oops I just saw that there may be a similar issue here too ... https://github.com/f4exb/sdrangel/issues/1717

srcejon, 2023-06-29:
#1717 is probably only relevant if you did a git pull today. So you might want to rollback to yesterday's code and see if there's a difference.

Valstee, 2023-06-29:
For my 1st problem, adding the Qt libraries (`qttools5-dev qttools5-dev-tools libspeexdsp-dev libsamplerate0-dev`) seems to have stopped the problem, I can't recreate this anymore : 

> (W) AudioDeviceManager::setOutputDeviceInfo: index: -1 device: System default device no FIFO registered yet hence no audio output has been allocated yet

f4exb, 2023-06-29:
`(W) PluginManager::loadPluginsDir: Cannot load library /opt/install/sdrangel/lib/sdrangel/pluginssrv/liblocalsinksrv.so: (/opt/install/sdrangel/lib/sdrangel/pluginssrv/liblocalsinksrv.so: undefined symbol: _ZN17LocalSinkSettings13m_maxFFTBandsE)`  it seems some compilers do not recognize static values allocation in .h files

Valstee, 2023-06-29:
>  So you might want to rollback to yesterday's code and see if there's a difference.

I've just installed SDRangelSrv v7.14.2 and I still have this problem: 
```

2023-06-29 18:54:29.674 (D) WFMDemodSink::feed: 0/16384 audio samples written
2023-06-29 18:54:30.018 (C) AudioFifo::write: (WFMDemod [0:0]) overflow 16384 samples

```

Valstee, 2023-06-30:
So it wasn't coming from SDRangel, I used the following commands to reinstall pavucontrol and the pulse audio config file and it works perfectly again : 

```
sudo apt remove --purge pavucontrol -y && sudo apt install pavucontrol -y
rm -r ~/.config/pulse/
```