f4exb/sdrangel #1725: Question: Is there a possibility of fetching the satellite pass data from the SatelliteTracker plugin?

Adeel-Abdullah, 2023-06-30:
The satellitetracker plugin calculates satellite pass data and then displays them in a neat little table, I was wondering if it's possible to fetch this somehow through an api or some other means. Accessing them directly through the plugin would be better than calculating them again seperately.

srcejon, 2023-06-30:
Yes, you should be able to use the swagger API

https://github.com/f4exb/sdrangel/wiki/Web-(http)-REST-API

srcejon, 2023-07-01:
Try:

    curl -X GET "http://127.0.0.1:8091/sdrangel/featureset/feature/0/report" -H  "accept: application/json"

Adeel-Abdullah, 2023-07-01:
Thank you, this is exactly what I needed! I somehow missed this when I was going through the API earlier.

closed 2023-07-03 by srcejon