f4exb/sdrangel #1726: I cannot find the ATV or Analogue TV option

MaxDeepCoder, 2023-07-02:
I cannot find the ATV or Analogue TV option in the latest binary.

Please help?

srcejon, 2023-07-02:
What OS?

MaxDeepCoder, 2023-07-02:
Windows 11

MaxDeepCoder, 2023-07-02:
I reinstalled it aswell.

MaxDeepCoder, 2023-07-02:
[image]

MaxDeepCoder, 2023-07-02:
There is no plugin called modatv.dll

f4exb, 2023-07-03:
The ATV modulator has a dependency on OpenCV: https://github.com/f4exb/sdrangel/blob/master/plugins/channeltx/CMakeLists.txt#L51

srcejon, 2023-07-03:
Yep - it's missing from the release build.

If I look at the GitHub build log for PR checks (E.g. https://github.com/f4exb/sdrangel/pull/1718/checks), I see

CMake Warning at external/windows/opencv/OpenCVConfig.cmake:166 (message):
  Found OpenCV Windows Pack but it has no binaries compatible with your
  configuration.

  You should manually point CMake variable OpenCV_DIR to your build of OpenCV
  library.
Call Stack (most recent call first):
  CMakeLists.txt:692 (find_package)

CMake Warning at CMakeLists.txt:692 (find_package):
  Found package configuration file:

    D:/a/sdrangel/sdrangel/external/windows/opencv/OpenCVConfig.cmake

  but it set OpenCV_FOUND to FALSE so package "OpenCV" is considered to be
  NOT FOUND.

srcejon, 2023-07-03:
From the log, CMake sets MSVC_VERSION to 1936

However, in OpenCVConfig.make, only 191* and 192* appear to be supported.

I.e. seems to not support Visual Studio 2022.

srcejon, 2023-07-03:
Have added a PR for sdrangel-windows-libraries with a quick fix for VS2022 using vc15 libs.

Perhaps we could do with a bit of a tidy up there though, as we have opencv (vc14,vc15) and opencv4 (v16) directories, but we're now building using vc17.

closed 2023-07-03 by f4exb