f4exb/sdrangel #1729: SDRAngel don't start under Linux Mint 21 Vanessa

DG5VE, 2023-07-04:
Hello, just now I installed V 7.15.0 but it won't start after sucessful installation. Here now the copy of all console messages:

dg5ve@Algol:~$ sdrangel
2023-07-04 10:45:43.933 (I) SDRangel 7.15.0 Qt 5.15.3 64b x86_64 Linux Mint 21 DSP Rx:24b Tx:16b PID 23660
2023-07-04 10:45:43.934 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/dg5ve/.config/f4exb/SDRangel.conf
2023-07-04 10:45:43.934 (D) MessagePipes::startGC
2023-07-04 10:45:43.934 (D) DataPipes::startGC
2023-07-04 10:45:43.939 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #0: default
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #1: samplerate
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #2: speexrate
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #3: jack
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #4: oss
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #5: pulse
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #6: upmix
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #7: vdownmix
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #8: hw:CARD=PCH,DEV=0
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #9: plughw:CARD=PCH,DEV=0
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #10: sysdefault:CARD=PCH
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #11: front:CARD=PCH,DEV=0
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #12: dsnoop:CARD=PCH,DEV=0
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #13: usbstream:CARD=PCH
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #14: alsa_input.pci-0000_00_1b.0.analog-stereo
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #15: alsa_output.pci-0000_00_1b.0.analog-stereo.monitor
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #16: PulseEffects_apps.monitor
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: input device #17: PulseEffects_mic.monitor
2023-07-04 10:45:44.193 (D) AudioDeviceManager::AudioDeviceManager: scan output devices
2023-07-04 10:45:44.667 (D) AudioDeviceManager::AudioDeviceManager: output device #0: default
2023-07-04 10:45:44.667 (D) AudioDeviceManager::AudioDeviceManager: output device #1: samplerate
2023-07-04 10:45:44.667 (D) AudioDeviceManager::AudioDeviceManager: output device #2: speexrate
2023-07-04 10:45:44.667 (D) AudioDeviceManager::AudioDeviceManager: output device #3: jack
2023-07-04 10:45:44.667 (D) AudioDeviceManager::AudioDeviceManager: output device #4: oss
2023-07-04 10:45:44.667 (D) AudioDeviceManager::AudioDeviceManager: output device #5: pulse
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #6: upmix
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #7: vdownmix
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #8: hw:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #9: hw:CARD=PCH,DEV=3
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #10: hw:CARD=PCH,DEV=7
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #11: plughw:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #12: plughw:CARD=PCH,DEV=3
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #13: plughw:CARD=PCH,DEV=7
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #14: sysdefault:CARD=PCH
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #15: front:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #16: surround21:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #17: surround40:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #18: surround41:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #19: surround50:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #20: surround51:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #21: surround71:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #22: hdmi:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #23: hdmi:CARD=PCH,DEV=1
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #24: dmix:CARD=PCH,DEV=0
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #25: dmix:CARD=PCH,DEV=3
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #26: dmix:CARD=PCH,DEV=7
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #27: usbstream:CARD=PCH
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #28: alsa_output.pci-0000_00_1b.0.analog-stereo
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #29: PulseEffects_apps
2023-07-04 10:45:44.668 (D) AudioDeviceManager::AudioDeviceManager: output device #30: PulseEffects_mic
2023-07-04 10:45:44.668 (D) MainWindow::MainWindow: start
2023-07-04 10:45:44.774 (D) FFTEngine::create: using FFTW engine
2023-07-04 10:45:44.775 (D) FFTWEngine::configure: no FFTW wisdom file
2023-07-04 10:45:44.878 (D) FFT: creating FFTW plan (n=128,forward) took 102 ms
2023-07-04 10:45:44.878 (D) FFTEngine::create: using FFTW engine
2023-07-04 10:45:44.878 (D) FFTWEngine::configure: no FFTW wisdom file
2023-07-04 10:45:45.085 (D) FFT: creating FFTW plan (n=256,forward) took 207 ms
2023-07-04 10:45:45.085 (D) FFTEngine::create: using FFTW engine
2023-07-04 10:45:45.085 (D) FFTWEngine::configure: no FFTW wisdom file
2023-07-04 10:45:45.450 (D) FFT: creating FFTW plan (n=512,forward) took 364 ms
2023-07-04 10:45:45.450 (D) FFTEngine::create: using FFTW engine
2023-07-04 10:45:45.450 (D) FFTWEngine::configure: no FFTW wisdom file
[... 369 lines cut ...]
2023-07-04 10:45:48.434 (D) DeviceUSRPParams::open: m_nbRxChannels:  1  m_nbTxChannels:  1
2023-07-04 10:45:48.434 (D) DeviceUSRP::enumOriginDevices: found USRP device  "N/A[3:$1] 2654740995182997645"
2023-07-04 10:45:48.434 (D) DeviceUSRPParams::open: default_input=False,default_output=False,device_id=5,driver=audio,label=Monitor of PulseEffects(mic),type=soapy,serial=2654740995182997645
2023-07-04 10:45:48.451 (D) DeviceUSRPParams::open: m_nbRxChannels:  1  m_nbTxChannels:  1
2023-07-04 10:45:48.451 (D) DeviceUSRP::enumOriginDevices: found USRP device  "N/A[4:$1] 000000000000000075b068dc32496707"
2023-07-04 10:45:48.451 (D) DeviceUSRPParams::open: device=HackRF One,driver=hackrf,label=HackRF One #0 75b068dc32496707,part_id=a000cb3c0081475d,serial=000000000000000075b068dc32496707,version=2018.01.1,type=soapy
[INFO] [UHDSoapyDevice] Opening HackRF One #0 75b068dc32496707...
2023-07-04 10:45:49.501 (D) DeviceUSRPParams::open: m_nbRxChannels:  1  m_nbTxChannels:  1
2023-07-04 10:45:50.012 (D) USRPInputPlugin::enumSampleSources: device #0 channel 0: default_input=True,default_output=True,device_id=0,driver=audio,label=Eingebautes Tongerät Analog Stereo,type=soapy,serial=4243088534115653241
2023-07-04 10:45:50.012 (D) USRPInputPlugin::enumSampleSources: device #1 channel 0: default_input=False,default_output=False,device_id=3,driver=audio,label=Monitor of Eingebautes Tongerät Analog Stereo,type=soapy,serial=5526514629394919524
2023-07-04 10:45:50.012 (D) USRPInputPlugin::enumSampleSources: device #2 channel 0: default_input=False,default_output=False,device_id=4,driver=audio,label=Monitor of PulseEffects(apps),type=soapy,serial=17050099503517592352
[... cut ...]
2023-07-04 10:45:50.012 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesink.bladerf1output
2023-07-04 10:45:50.016 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesink.bladerf2output
2023-07-04 10:45:50.019 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesink.fileoutput
2023-07-04 10:45:50.019 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesink.hackrf
Speicherzugriffsfehler (Speicherabzug geschrieben)
dg5ve@Algol:~$ 

Linux Mint 21 Vanessa is fresh updated and works well.

Thank you for any help and best 73 from Sigmar, DG5VE

f4exb, 2023-07-04:
It would be useful to have the stack trace of this core dump (Speicherabzug)

DG5VE, 2023-07-04:
Where can I find it exactly?

f4exb, 2023-07-04:
The sdrangel program that produced the core dump is needed as well

f4exb, 2023-07-04:
alternatively you can use gdb -c <core> sdrangel and then get the stack trace with bt command and paste it here

DG5VE, 2023-07-05:
> The sdrangel program that produced the core dump is needed as well

so you need the binary file? Here it is.
[binaries.tar.gz](https://github.com/f4exb/sdrangel/files/11955457/binaries.tar.gz)

DG5VE, 2023-07-16:
Hello,
do you maybe need any further informations? Maybe I should try any previous version again.
But modtly I need to pre-install some older libraries separately. I feel not happy with that because
that seems to cost some stability of my Linux system.

DG5VE, 2023-07-16:
Just now I tried out the new version...sadly still the same problem...

f4exb, 2023-07-17:
The core file is not exploitable. I don't know what happens sorry.

closed 2023-07-17 by f4exb

DG5VE, 2023-07-17:
So it was useless to ask for any soution. Thanks for nothing. Was a mistake to concentrate to this software.

f4exb, 2023-07-17:
I am sorry but when describing a problem you should give sufficient material. I still cannot get a stack trace. With minimal knowledge of Linux and gdb this should be easy to obtain on your side and I cannot do it for you. For the vast majority of users this is working so it must be a corner case in your environment that I cannot reproduce for you. 

Now just considering the log. It seems to break at device enumeration time so what I can suggest is removing all connected devices and start the program and see if it fails (hopefully not) then reconnect each device one at a time to be able to see which one causes the fault. Then you have to investigate what is wrong with this device and I cannot do it for you.