f4exb/sdrangel #1734: SIGABRT in ADSBDemodSinkWo

dforsi, 2023-07-08:
sdrangel is terminated when closing the window of the ADS-B Demodulator after it received some samples, even just noise.

Steps to reproduce:
* Create a new workspace
* Add Rx Device: Audio
* Add channels: ADS-B Demodulator
* start acquisition (the settings don't matter)
* close the window of the ADS-B Demodulator
* SIGABRT

```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Thread 126 "ADSBDemodSinkWo" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffeb50ed6c0 (LWP 59960)]
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: File o directory non esistente.
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x00007ffff5aa9d2f in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x00007ffff5a5aef2 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff5a45472 in __GI_abort () at ./stdlib/abort.c:79
#4  0x00007ffff589d919 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff58a8e1a in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff58a8e85 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff5c8ffcd in qTerminate() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff5c91a02 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff5aa7fd4 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#10 0x00007ffff5b285bc in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

f4exb, 2023-07-09:
> Add Rx Device: Audio

You need at least 2 MS/s for the ADS-B demod to work correctly anyway and I don't know of any soundcard capable of this.

dforsi, 2023-07-09:
The same SIGABORT is triggered when choosing an SDR capable of delivering 2 MS/s; both before decoding any packet and after decoding some, closing the "ADS-B Demodulator" window with the "x" button terminates the whole application.

srcejon, 2023-07-09:
Looks like a problem if the ADSB window is closed while acquisition is running. Ok, if not running. Crashes on Linux, but not on Windows.

closed 2024-11-12 by f4exb