f4exb/sdrangel #1735: Sdrangel vs Ubuntu 23 04

choupette-change, 2023-07-12:
Question: Have you tested Sdrangel vs Ubuntu 23 04?

robbietee, 2023-07-14:
Yes. Compiled from source on Ubuntu 23.04. Spectrum windows show the window or desktop behind as if the window is transparent then it gets progressively weirder and not usable.
[image]
[image]

robbietee, 2023-07-15:
Screenshots didn't catch the issue correctly. The actual screen looks more like this:
[image]
[image]

robbietee, 2023-07-15:
Graphics issues aside, functionality is good. I found no issues with:
Rtl-Sdr 

Channels:
ADS-B
AIS
AM
Broadcast FM
DAB
NFM
Packet
Pager
Radiosonde
Packet
Pager

Features:
AIS
APRS
Map
Radiosonde

I've never had much luck with digital modes on any version. A "Configurations Download" Page would be useful.

Thanks, Rob M6RXT

f4exb, 2023-07-17:
Not an issue with Ubuntu 23.04 nor the code. It compiles and wirks:
[image]

srcejon, 2023-07-17:
I see the same problem,  so presumably something graphics driver / opengl related.

f4exb, 2023-07-18:
Make sure you start compiling from scratch. The drivers have probably been upgraded this is to make sure the current version is used.

robbietee, 2023-07-20:
I compiled from scratch on a fresh install and still have the issue. I will start researching.....

 *-display
             description: VGA compatible controller
             product: 3rd Gen Core processor Graphics Controller
             vendor: Intel Corporation
             physical id: 2
             bus info: pci@0000:00:02.0
             logical name: /dev/fb0
             version: 09
             width: 64 bits
             clock: 33MHz
             capabilities: msi pm vga_controller bus_master cap_list rom fb
             configuration: depth=32 driver=i915 latency=0 resolution=1366,768
             resources: irq:31 memory:f6400000-f67fffff memory:e0000000-efffffff ioport:f000(size=64) memory:c0000-dffff

srcejon, 2023-07-31:
I just compiled the Qt openglwindow example (from Qt Creator) and this shows a similar corruption. So I suspect it's a Qt/Graphics driver issue.

robbietee, 2023-07-31:
It's so annoying. I don't know why I bothered with the 23.04 "upgrade". Having said that, I haven't found any other issues with it....yet.

srcejon, 2023-07-31:
If you can build from source, give that patch a try - it fixes it for me.

PR #1756 (2023-08-01): Clear OpenGL buffer with alpha=1
Fixes #1735 for me.
commit 37285a9b (2023-08-01): Clear with alpha=1 to fix #1735
closed 2023-08-01 by f4exb