f4exb/sdrangel #1736: Porting to the sBitx v2

rafael2k, 2023-07-13:
Any guide on how to port to another radio (1 channel alsa for signal, and some gpio commands for frequency), called sBitx.

f4exb, 2023-07-14:
If it can be controlled via CAT you can use the Audio CAT SISO plugin: https://github.com/f4exb/sdrangel/tree/master/plugins/samplemimo/audiocatsiso#audio-cat-siso-plugin

rafael2k, 2023-07-15:
Perfect. Will give a try and report my results. I'm running Linux on the Pi 4. I was not really planning to use hamlib (but call directly the radio functions using a shared memory interface I created [1]).

[1] https://github.com/Rhizomatica/hermes-net/tree/main/trx_v2-userland

closed 2023-08-31 by github-actions[bot]

rafael2k, 2023-11-15:
Hi Edouard, would it be ok if I reopen this one. I think the way to go is a specific plugin for supporting the sbitx, for example, to deal with using left and right for different purposes (mic and rx, tx and speaker), the passband is aroud 6 kHz up to around 30 kHz, and talk with the I/O daemon over shared memory as I'm writing in here:
https://github.com/Rhizomatica/hermes-net/tree/main/trx_v2-userland