f4exb/sdrangel #1737: DMG Please

loewal, 2023-07-13:
DMG Please....

ThnX

srcejon, 2023-07-14:
Not for a couple of weeks, I'm afraid,  as dont currently have access to a mac.

artscotty, 2023-07-20:
> Not for a couple of weeks, I'm afraid, as dont currently have access to a mac.

Thanks Jon , I managed to compile 7.15.1 from src on ubuntu but im not able to compile on Mac as I cant get it to recognise my libusb and it errors out too much. I look forward you you getting a DMG out there for the TX audio fix version! Thanks again

K7IU, 2023-07-29:
Is a .dmg for 7.15.1 available yet?  Is there a 7.14 .dmg someplace that I can try out until 7.15 is available?

srcejon, 2023-07-29:
No 7.15.1, but 7.15.0 is available on the releases page if you click the "Assests" under 7.15.0 button a list of files will be displayed.

K7IU, 2023-07-29:
Thanks, got it.Sent from my iPadAnd YES it's cool!

srcejon, 2023-08-04:
DMG built for 7.15.2.

closed 2023-08-04 by srcejon