f4exb/sdrangel #1740: DAB demodulator plugin missing in Linux deb package for version 7.15.1

computer-wizard, 2023-07-15:
The dab demodulator plugin is missing in the Linux deb package for version 7.15.1.
I checked the /usr/lib/sdrangel/plugins sub-directory and the libdemod library for 
dab isn't there.

The only library related to dab is /usr/lib/sdrangel/libdab_lib.so but as I already said earlier,  
there is no libdemod library for it :-(

srcejon, 2023-07-28:
@f4exb - How can I see the generated CMakeCache.txt for the ubuntu2204prod build on appveyor to debug? Does it need to be added as an additional artifact in .appveyor.yml, or is is possible to ssh in?

In the ubuntu2204prod  log file, it seems the dab project is built, so I presume both ZLIB_FOUND AND FAAD_FOUND must be true, and that should set DAB_FOUND to TRUE.

The demoddab build is conditional on:

    if(ENABLE_CHANNELRX_DEMODDAB AND DAB_FOUND AND ZLIB_FOUND AND FAAD_FOUND)
        add_subdirectory(demoddab)
    endif()

So can't then see why it wouldn't then be built.

srcejon, 2023-07-30:
It looks like when AUTO_EXTERNAL_LIBRARIES is set in external/CMakeLists.txt, the FOUND variables aren't cached, so they are undefined in the plugins CMakeLists.txt

From the new log:

Not building demoddab (ENABLE_CHANNELRX_DEMODDAB=ON DAB_FOUND=ON ZLIB_FOUND= FAAD_FOUND=)
[00:23:06] -- Not building sigmffilesink (ENABLE_CHANNELRX_SIGMFFILESINK=ON LIBSIGMF_FOUND=)
[00:23:06] -- Not building modatv (ENABLE_CHANNELTX_MODATV=ON OpenCV_FOUND=)

computer-wizard, 2023-07-30:
Good find. I wonder why they aren't cached?

f4exb, 2023-07-30:
Appveyor also has its own build cache. I just cleaned it.

srcejon, 2023-07-30:
> Good find. I wonder why they aren't cached?

I think it's a scope problem.

When AUTO_EXTERNAL_LIBRARIES is set, the find_package variables from external/CMakeLists.txt are lost at the end of that script.

Recent versions of cmake have a global flag to find_package, that would keep them in scope, but the more compatible solution, I think, is just to move the find_package commands in to the root CMakeLists.txt. So, I'll give that a try...

srcejon, 2023-07-30:
Dab Demod and ATV mod are now being built.

Still have a couple of problems though:

[00:24:40] -- Not building remotesink (ENABLE_CHANNELRX_REMOTESINK=ON CM256CC_FOUND=FALSE HAS_SSE3=1 HAS_NEON=)
[00:24:40] -- Not building sigmffilesink (ENABLE_CHANNELRX_SIGMFFILESINK=ON LIBSIGMF_FOUND=)
[00:24:40] -- Include demoddatv
[00:24:40] -- Not building remotesource (ENABLE_CHANNELTX_REMOTESOURCE=ON CM256CC_FOUND=FALSE HAS_SSE3=1 HAS_NEON=)

srcejon, 2023-07-30:
Hmm. It seems if CM256CC_FOUND is set by the call to find_package by the root script, when set(CM256CC_FOUND ON CACHE INTERNAL "") is called in external/CMakeLists.txt, this only sets the value in the cache, it doesn't update the variable in the parent scope, and so the value (FALSE) returned by find_package is used.

srcejon, 2023-07-30:
Sigmffilesink not being built, appears to be because libsigmf isn't available. So will try to build that in external/CMakeLists.txt

srcejon, 2023-07-31:
@computer-wizard you can try this build: https://ci.appveyor.com/api/buildjobs/bixaf8pfyi91j7p1/artifacts/sdrangel-2195-master.tar.gz

It includes these following additional libraries which were missing from the current release.

libfcdhid.so
libfcdlib.so
liblibsigmf.so
libdemoddab.so
libinputfcdproplus.so
libinputfcdpro.so
libinputsigmffileinput.so
libmodatv.so
libsigmffilesink.so

computer-wizard, 2023-07-31:
I will try it, many thanks!

computer-wizard, 2023-08-02:
DAB works, many thanks for fixing it

closed 2023-08-02 by computer-wizard