f4exb/sdrangel #1741: Problem to compil Codec2/FreeDV

choupette-change, 2023-07-20:
I have a problem to compil Codec2/FreeDV, I try some times
tutu@ccccccccccc:/opt/build$ git clone https://github.com/drowe67/codec2.git
Clonage dans 'codec2'...
remote: Enumerating objects: 1143, done.
remote: Counting objects: 100% (1143/1143), done.
remote: Compressing objects: 100% (810/810), done.
remote: Total 1143 (delta 413), reused 1051 (delta 324), pack-reused 0
Réception d'objets: 100% (1143/1143), 15.24 Mio | 3.41 Mio/s, fait.
Résolution des deltas: 100% (413/413), fait.
tutu@tcccccccc:/opt/build$ cd codec2
tutu@tcccccccc:/opt/build/codec2$ git reset --hard 76a20416d715ee06f8b36a9953506876689a3bd2
**fatal: Impossible d'analyser l'objet '76a20416d715ee06f8b36a9953506876689a3bd2'.**

f4exb, 2023-07-21:
I guess the commit 76a20416d715ee06f8b36a9953506876689a3bd2 has been squashed. Sticking to it was probably due to a specific issue and it may not be longer valid. Just try without "git reset --hard ..."

f4exb, 2023-07-24:
It appears that all tags have been romoved:
```
$ git reset --hard "v1.0.3"
fatal: ambiguous argument 'v1.0.3': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
```

f4exb, 2023-07-26:
That was only 2 days ago.

f4exb, 2023-07-27:
Wiki updated

closed 2023-07-27 by f4exb

srcejon, 2023-07-29:
Looks like a similar change needed in cmake/ci/build_codec2.sh - I think that's why the docker build is failing

srcejon, 2023-07-29:
Nope, that's not write. Looks like it's the docker script that needs fixing: https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/Dockerfile