f4exb/sdrangel #1749: Inconsistent "codec2" and "codec2-dev" in "Compile from source in Linux" webpage.

gvimlag, 2023-07-30:
In "https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux" under "Codec2/FreeDV":

```
sudo apt-get -y install libspeexdsp-dev libsamplerate0-dev
cd /opt/build
git clone https://github.com/drowe67/codec2-dev.git
cd codec2
git reset --hard "v1.0.3"
mkdir build_linux; cd build_linux
cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/codec2 ..
make -j $(nproc) install
```

The above commands clone "https://github.com/drowe67/codec2-dev.git",
but then try to cd into "codec2", not "codec2-dev".

Both repositories are available:
  https://github.com/drowe67/codec2.git
  https://github.com/drowe67/codec2-dev.git

Which is correct?

srcejon, 2023-07-30:
Try checking out codec2-dev into codec2 directory, by adding dir name at the end of the clone command:

    git clone https://github.com/drowe67/codec2-dev.git codec2

f4exb, 2023-07-30:
Wiki updated with the command above

closed 2023-07-30 by f4exb