f4exb/sdrangel #1752: DSD (Digital Speech Decoder) not working - garbled or no audio

DavidBeavan, 2023-07-30:
First of all, thanks for the great software, and for making it open.

I'm having unworkable DSD decodes, it works with 6.18.1 (not scientifically chosen). It looks like it may be related to #1592  or #1637 opening this issue as both of those are closed.

**Comparing to working it exhibits:**
* Spectrum scope blank or seems to update infrequently
* Garbled or no audio

**For me, I can reproduce**
* IQ file (as @cborss [also used](https://github.com/f4exb/sdrangel/issues/1592#issuecomment-1644538018)) https://sdrplay.com/resources/IQ/dmr.zip
* SDRAngel 7.15.1 deb
* Ubuntu 22.04
* Inside VirtualBox
* Windows 10 host
* SDRPlay Duo (but same results if I stop SDRPlay API and have it unplugged)

I'm very happy to do more investigation, if you let me know what you might need?

Working (6.18.1)
[image]

Not working (7.15.1)
[image]

srcejon, 2023-07-30:
Thanks for the good bug report, which I can reproduce. The problem seems to appear between 6.20.1 and 7.0.0.

cborss, 2023-07-30:
I also did some further testing and generated some debug output (see attachment). I'm using SDRangel 7.15.1 and 6.20.3 which I compiled from source code. In the "_cleaned.log" variants I replaced the timestamps to make it easier to compare them (with meld or any other diff tool).

The DSDDecoder log messages look actually pretty similar. Hence, I thought it might be worth a try to dump the audio buffers at the end of DSDDemodSink::feed() into a file to check if maybe the audio pipeline is no longer working properly. But I got the same mostly muted messed up audio signal. So the bug seems to be before that point.

Just wanted to share this information with you. Hope it helps to narrow down the issue.

[debug_log.zip](https://github.com/f4exb/sdrangel/files/12209869/debug_log.zip)

srcejon, 2023-08-01:
It seems to go wrong between these two commts (which are for the v7 UI)

38c1a64b71ed25b9da251c0dfecd72239db00e62  - apr 4 2022 - OK
2f89b79c8491a970c9208a616e1599ac4e2c10ba - apr 12 2022 - Not working

There are a few intermediate commits, but the GUI is not functional, so can't test.

I can't actually see any changes to the DSD Demod between these two reviews, that indicate where the problem might have been introduced - so it might be something in sdrbase/sdrgui that is responsible.

srcejon, 2023-08-01:
Actually, looks like a simple GUI issue, as the Slot 2 On/Off button wasn't connected properly in the V7 GUI. Above patch fixes it for me.

cborss, 2023-08-01:
That was quick. Thank you!
I just tested your patch and it works for me. :)

DavidBeavan, 2023-08-02:
Looks like great news, I'm a bit preoccupied with other tasks, so cant check out any changes for a while. Which means, don't wait for me, if you think it's fixed :)

f4exb, 2023-08-04:
Deployed in v7.15.2

closed 2023-08-04 by f4exb