f4exb/sdrangel #1755: Swagger codegen documentation stale link

lrowen, 2023-07-31:
At [https://github.com/f4exb/sdrangel/tree/master/swagger/sdrangel#installation](url) a link is given to a fixed version of the swagger qt5 code generator.  This link does not work as of 7/31/2023.  For those who are just starting to get introduced to swagger and how it's integrated in SDRangel, it would be good to have a correct link or correct information here.

f4exb, 2023-07-31:
Which link?

f4exb, 2023-07-31:
This is what you should be looking for: https://github.com/f4exb/sdrangel/tree/master/swagger/docker

lrowen, 2023-07-31:
The link in the page I referenced is [https://github.com/etherealjoy/swagger-codegen/tree/qt5cpp_rework_antis81_patch-1](url) which throws a 404.

Sorry; I just right clicked the link icon in the listing I referenced in the link; the actual file with the link is the README.md in that directory.  The link into the file itself, as given by the github web interface, is [https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/README.md#installation](url)

FWIW I'm not using docker.  I understand much of the information is usable outside docker, but I was just giving a 'link no longer available' report for something linked in one of the README.md files.

f4exb, 2023-08-01:
It is much easier to go the Docker way if you want to regenerate the Swagger files.

lrowen, 2023-08-01:
Ok, thanks. Closing.

closed 2023-08-01 by lrowen