f4exb/sdrangel #1757: Mac OS ATV plugin doesn't work

no111u3, 2023-07-31:
Than I start using ATV plugin it has black screen on video tab and error in console:
```
2023-08-01 00:32:47.751 (W) QOpenGLShader::compile(Fragment): ERROR: 0:13: Invalid call of undeclared identifier 'texture2D'
ERROR: 0:14: Invalid call of undeclared identifier 'texture2D'
ERROR: 0:15: Use of undeclared identifier 'tshift1'
ERROR: 0:16: Use of undeclared identifier 'tshift3'
ERROR: 0:17: Use of undeclared identifier 'shift1'
ERROR: 0:18: Use of undeclared identifier 'shift3'
ERROR: 0:19: Use of undeclared identifier 'txs1'
ERROR: 0:20: Use of undeclared identifier 'txs3'
ERROR: 0:21: Use of undeclared identifier 'p1x'
ERROR: 0:22: Use of undeclared identifier 'p3x'
ERROR: 0:23: Use of undeclared identifier 'p1x'
ERROR: 0:24: Use of undeclared identifier 'p2x'
ERROR: 0:25: Use of undeclared identifier 'p3x'
ERROR: 0:26: Use of undeclared identifier 'p4x'
ERROR: 0:27: Use of undeclared identifier 'p1'
ERROR: 0:27: Use of undeclared identifier 'p2'
ERROR: 0:27: Use of undeclared identifier 'txs1'
ERROR: 0:28: Use of undeclared identifier 'p3'
ERROR: 0:28: Use of undeclared identifier 'p4'
ERROR: 0:28: Use of undeclared identifier 'txs3'
ERROR: 0:29: Use of undeclared identifier 'p12'
ERROR: 0:29: Use of undeclared identifier 'p34'
ERROR: 0:30: Use of undeclared identifier 'p'

2023-08-01 00:32:47.751 (W) *** Problematic Fragment shader source code ***
#version 330
#ifdef GL_KHR_blend_equation_advanced
#extension GL_ARB_fragment_coord_conventions : enable
#extension GL_KHR_blend_equation_advanced : enable
#endif
#define lowp
#define mediump
#define highp
#line 1
uniform highp sampler2D tex1;
uniform highp sampler2D tex2;
uniform highp float imw;
uniform highp float imh;
uniform highp float tlw;
uniform highp float tlh;
in highp vec2 texCoordVar;
out vec4 fragColor;
void main() {
    float tlhw = 0.5 * tlw;    float tlhh = 0.5 * tlh;    float tys = (texCoordVar.y + tlhh) * imh;
    float p1y = floor(tys) * tlh - tlhh;
    float p3y = p1y + tlh;
    float tshift1 = texture2D(tex2, vec2(0.0, p1y)).r;
    float tshift3 = texture2D(tex2, vec2(0.0, p3y)).r;
    float shift1 = (1.0 - tshift1 * 2.0) * tlw;
    float shift3 = (1.0 - tshift3 * 2.0) * tlw;
    float txs1 = (texCoordVar.x + shift1 + tlhw) * imw;
    float txs3 = (texCoordVar.x + shift3 + tlhw) * imw;
    float p1x = floor(txs1) * tlw - tlhw;
    float p3x = floor(txs3) * tlw - tlhw;
    float p2x = p1x + tlw;
    float p4x = p3x + tlw;
    float p1 = texture(tex1, vec2(p1x, p1y)).r;
    float p2 = texture(tex1, vec2(p2x, p1y)).r;
    float p3 = texture(tex1, vec2(p3x, p3y)).r;
    float p4 = texture(tex1, vec2(p4x, p3y)).r;
    float p12 = mix(p1, p2, fract(txs1));
    float p34 = mix(p3, p4, fract(txs3));
    float p = mix(p12, p34, fract(tys));
    fragColor = vec4(p);
}

***
2023-08-01 00:32:47.751 (W) TVScreenAnalog::initializeGL: error in fragment shader: "ERROR: 0:13: Invalid call of undeclared identifier 'texture2D'\nERROR: 0:14: Invalid call of undeclared identifier 'texture2D'\nERROR: 0:15: Use of undeclared identifier 'tshift1'\nERROR: 0:16: Use of undeclared identifier 'tshift3'\nERROR: 0:17: Use of undeclared identifier 'shift1'\nERROR: 0:18: Use of undeclared identifier 'shift3'\nERROR: 0:19: Use of undeclared identifier 'txs1'\nERROR: 0:20: Use of undeclared identifier 'txs3'\nERROR: 0:21: Use of undeclared identifier 'p1x'\nERROR: 0:22: Use of undeclared identifier 'p3x'\nERROR: 0:23: Use of undeclared identifier 'p1x'\nERROR: 0:24: Use of undeclared identifier 'p2x'\nERROR: 0:25: Use of undeclared identifier 'p3x'\nERROR: 0:26: Use of undeclared identifier 'p4x'\nERROR: 0:27: Use of undeclared identifier 'p1'\nERROR: 0:27: Use of undeclared identifier 'p2'\nERROR: 0:27: Use of undeclared identifier 'txs1'\nERROR: 0:28: Use of undeclared identifier 'p3'\nERROR: 0:28: Use of undeclared identifier 'p4'\nERROR: 0:28: Use of undeclared identifier 'txs3'\nERROR: 0:29: Use of undeclared identifier 'p12'\nERROR: 0:29: Use of undeclared identifier 'p34'\nERROR: 0:30: Use of undeclared identifier 'p'\n"
```

srcejon, 2023-08-01:
fragmentShaderSource in tvscreenanalog.cpp should use texture() instead of texture2d().

no111u3, 2023-08-01:
Fixed
[image]

closed 2023-08-01 by no111u3

AlexeyBarsuk, 2025-06-28:
Texture2D is used multiple times in master branch, that makes some modules (spectrogram, atv) not to work on macosx with error 
```
QOpenGLShader::compile(Fragment): ERROR: 0:13: Invalid call of undeclared identifier 'texture2D'
```

[image]
[image]

srcejon, 2025-06-28:
There are two versions of those shades though. The newer version just uses texture. So the question is why are you getting the old version. Can you post your log file?

AlexeyBarsuk, 2025-06-28:
This happens with sdrangel installed using .dmg and using macports both.
[sdrangel.log](https://github.com/user-attachments/files/20963801/sdrangel.log)

AlexeyBarsuk, 2025-06-29:
my apologies