f4exb/sdrangel #1760: File Sink can't save to user folders on Windows

CoolNamesAllTaken, 2023-08-04:
## Version Info
SDR Angel 7.15.2
Windows 11 Pro Version 22H2

## Problem Description
I noticed a strange behavior with the File Sink plugin on my machine, which runs Windows 11. My user account was created as `C:\Users\firstname.lastname`, so for instance my downloads folder is `C:\Users\firstname.lastname\Downloads`. Unfortunately, it looks like SDR Angel is splitting the file extension using the first instance of a `.` character instead of the last instance of a `.` character, so when I try to save a file to `C:\Users\firstname.lastname\Downloads\testfile.sdriq`, I get an error that SDR Angel is unable to open the file `C:\Users\firstname.2023-08-04T23_27_22_906.sdriq`, which is of course in a nonexistent directory.

I have gotten around this issue for now by recording everything directly to my `C:\` drive, but it would be really nice if I was able to record files directly to folders associated with my user account!

CoolNamesAllTaken, 2023-08-04:
My best guess is that the issue starts at the filename split somewhere around [here](https://github.com/f4exb/sdrangel/blob/83834674c0c604bbc67b9c59172e05eda87f907a/plugins/channelrx/filesink/filesinksink.cpp#L295).

srcejon, 2023-08-23:
Implemented in 7.15.3

closed 2023-08-23 by srcejon