f4exb/sdrangel #1768: APRS decoder has bug when decoding packets using the 'secondary' icon table

rogerclarkmelbourne, 2023-08-08:
Longitude value is not displayed when the APRS data uses the secondary icon table
Only Latitude is shown

I've not looked at the code for the APRS decoder, but the delimiter between Lat and Lon sections is  '/' for the primary table and '\' (backslash) , for the secondard table

It appears that the decoder is only checking for the primary table delimiter in the section that displays the Lat/Lon.

Strangely the icons display OK for the secondary table

I have confirmed the APRS data transmission is OK by transmitting via a repeater connected to APRS.fi, and the location and icon are displayed correctly

srcejon, 2023-08-09:
Do you have an IQ recording showing the problem?

rogerclarkmelbourne, 2023-08-09:
No.

I'll try to make a recording or copy and paste the raw bytes from the packet decoder

rogerclarkmelbourne, 2023-08-11:
I can no longer reproduce the bug.

So I'll close this, until I see the bug again

closed 2023-08-11 by rogerclarkmelbourne