f4exb/sdrangel #1773: Crash APT decoder

choupette-change, 2023-08-10:
I installed the 3 satellites on the same workspace to be able to monitor them more easily. When starting to decode the first of the satellites (e.g. NOAA 18 in the screenshot) SDRangel crashes?? Is this normal operation or a bug?
[image]
/home/tutu/sdrangel.log
[sdrangel.log](https://github.com/f4exb/sdrangel/files/12312222/sdrangel.log)

srcejon, 2023-08-10:
If it crashes, it's a bug.

srcejon, 2023-08-17:
I thought this used to work, but looking at the aptdec code, it doesn't appear to be reentrant (lots of static variables), so it's not going to work, unless aptdec is rewritten.