f4exb/sdrangel #1782: I hope RX888 support is being considered.

mistergitj, 2023-08-17:
Is supporting SDR RX888 being considered? Here are some links - good and not so good.
https://youtu.be/WmaB4dIh99Y
https://docs.google.com/document/d/1hG634yVkBBIMykmmnosUxw7gumAoc3z1SmWDrH1TucM/edit
Thanks, John.

f4exb, 2023-08-19:
Considering the document in the last link you mention sorry probably not.

mistergitj, 2023-08-19:
Thanks for your response, Edouard. I suspected that article might deter you but hoped you would look past it. There are lots of positive comments on the RX888. I have one on order to see what Direct Sampling is about. A request came right after mine (Add support for RTL-SDR Blog V4) that looks like a similar device without direct sampling. I hope you will reconsider supporting the RX888.
Thanks much, John.

nmaster2042, 2023-08-20:
@mistergitj you can watch this: https://github.com/cozycactus/SoapyRX888

It's an attempt to develop a soapySDR driver for this device.

I don't know if it will go to final stage.

The author is working on a lib rx888 https://github.com/cozycactus/librx888 for his SoapySDR Driver.

mistergitj, 2023-08-21:
Thanks, nmaster2042. I am a Windows user and know almost nothing about Linux. I have used GQRX many years ago but prefer to  use W10 and SDRangel for my radios. Thanks, John.

f4exb, 2023-08-22:
There is also this one: https://github.com/modrzywolski/rx666m

closed 2023-10-09 by github-actions[bot]