f4exb/sdrangel #1783: Add support for RTL-SDR Blog V4

rtlsdrblog, 2023-08-19:
Hi, I've just released the RTL-SDR Blog V4 (https://www.rtl-sdr.com/rtl-sdr-blog-v4-dongle-initial-release/) which is based on the R828D tuner chip. But the hardware is slightly different as it uses a different LO frequency for the R828D compared to what is hardcoded in the drivers, and the RF inputs need to be switched in a different way.

I've created a fork of the Osmocom drivers here https://github.com/rtlsdrblog/rtl-sdr-blog-minimal, and a more heavily modified one here with other unrelated changes https://github.com/rtlsdrblog/rtl-sdr-blog. These repo's were modified to add support for the V4, and the -minimal fork has the minimal amount of changed to make it work. I've submitted a patch to Osmocom to try and upstream the main changes for the V4, but the Osmocom rtl-sdr project might be dead as I see that patch requests have been unanswered for a few years.

If it's not upstreamed is there some other way to support the RTL-SDR Blog V4?

On Windows it's easy as I can just tell users to replace the .dll file, and on Linux users can just update the system RTL-SDR driver. But for MacOS and Android the driver gets shipped with the package, and it's a bit harder to update.

The forked drivers continue to support older RTL-SDRs and other brands/generic models too, nothing has been removed.

nmaster2042, 2023-08-19:
I think V4 will be usable when building SDR++ from source, building your fork before of the rtlsdr lib before.
If the fork exposes the same functions and settings, it will work, like changing DLL on windows.
I bought a V4 so I'll see if it's actually working like I think.

rtlsdrblog, 2023-08-19:
Compilation from source will work, but I would guess a lot of people are using the MacOS and Android packages which are prepackaged with the older driver.

srcejon, 2023-08-19:
I ordered one the other day, so will take a look when it arrives.

rtlsdrblog, 2023-08-22:
Just noting that Osmocom replied to my upstream request so it's likely that it will be upstreamed soon. So probably the only changes required will be pulling the Osmocom code when its patched, and updating the packages.

alphafox02, 2023-08-25:
I have one here now. Just tried with SDRAngel and as rtlsdrblog noted, needs updated in SDRAngel. 

Outside of SDRAngel it was as simple as pulling the latest osmocom rtlsdr w/ it's changes now and making a deb package for 22.04. I then sudo cp /usr/lib/x86_64-linux-gnu/librtlsdr.* /usr/lib/sdrangel and restarting sdrangel works great.

srcejon, 2023-08-26:
Patch updates RTLSDR driver and sets lower frequency limit to 0, to support the HF upsampler. Anything else needed?

rtlsdrblog, 2023-08-28:
Yep that's all that's needed, thanks!

srcejon, 2023-09-04:
Released in 7.15.4

closed 2023-09-04 by srcejon