f4exb/sdrangel #1784: Pointer to help file for USB Demodulation Plugin.

mistergitj, 2023-08-21:
I cannot find a help file (? icon) for the SSB demodulator. I hope someone will help me. Thanks, John.

srcejon, 2023-08-23:
https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodssb/readme.md

mistergitj, 2023-08-23:
Sorry, srcejon. I was premature. Here is an image of my SSB demodulator. It is not the same as the one you posted. Enjoy, John.

[image]

srcejon, 2023-08-23:
It's the same, just the picture in the docs is slightly out of date.

What specific question do you have/

mistergitj, 2023-08-23:
Thanks, srcejon. I will try to come up with some specific questions.  Enjoy, John.

closed 2023-10-09 by github-actions[bot]