f4exb/sdrangel #1787: Add support to Airspy mini

autodiag, 2023-08-23:
no detect airspy mini

srcejon, 2023-08-23:
Please include a log file (Preferences > Logging) and give us as much information as possible. What OS are you using? Does your mini work with other SDR s/w?

autodiag, 2023-08-24:
[sdrangel.log](https://github.com/f4exb/sdrangel/files/12427650/sdrangel.log)

system is W10x64
have Airspy mini 3pcs and one R2
R2 is detected
airspy nothing
all another work (linux &win x64)

autodiag, 2023-08-24:
now check same problem to sdrplay2

Airspy r2,Pluto,RTL-SDR  work

srcejon, 2023-08-24:
What happens if you just have a single mini connected (no r2).

For SDRplay, you need the SDRplay API installed - https://www.sdrplay.com/api/

autodiag, 2023-08-24:
my mistake, the log was without RSP2 connected, I will give you the log with it connected

yes of course I know what is yours, same situation as with airspy, everywhere else only SDR Angel works :/

and no other applications nor Airspy mini any RSP2

all tested after reboot

autodiag, 2023-08-24:
up

if there is only one Aispy it finds it, if there are more it ignores them and it doesn't matter what version of airspy it is

if RSP2 is connected together with Pluto sdrangel will remain on the splashscreen

if RSP2 is connected separately it is not detected anyway

info log include
[sdrangel.log](https://github.com/f4exb/sdrangel/files/12428965/sdrangel.log)

srcejon, 2023-08-24:
> if there is only one Aispy it finds it, if there are more it ignores them and it doesn't matter what version of airspy it is

Ok, have updated the issue title.
 
> if RSP2 is connected separately it is not detected anyway

In your log, you have:

2023-08-24 13:26:06.973 (C) SDRPlayV3Plugin::enumOriginDevices: Failed to get SDRPlay devices:  sdrplay_api_ServiceNotResponding

What version of SDRplay API do you have installed?
Is the SDRplay API windows service running? (Run Services program to check)
Does SDRplay's Uno work if you try it after getting this error in SDRangel?

srcejon, 2023-08-24:
Unfortunately I don't have several Airspy devices to test with. 

Do you have the same problem with SDRangel on Linux?

f4exb, 2023-08-25:
What are the serial numbers of your Airspy devices? You can have this issue if they all have the same serial number this is what would happen with RTL-SDR for example.

autodiag, 2023-08-27:
no, SN is diferent,its original no chnesse shit ;) ..

closed 2023-10-12 by github-actions[bot]