f4exb/sdrangel #1794: Pager decoder

choupette-change, 2023-08-28:
If we put several pager decoders (with "duplicate channel") on the same workspace and we adjust the frequency, the decoding of the messages of the other 2 "pager decoders", in my case, seem capricious?? What am I doing wrong? The signals decoded in "Waves forms" are ok for the 3 frequencies??
Best regards
[image]

srcejon, 2023-08-29:
Can you decode each channel ok just using a single demod?

choupette-change, 2023-08-29:
Yes, if I take the decoders one by one it works. I close and reopen Sdrangel between each frequency test.

[image]

srcejon, 2023-08-29:
Ok - can you please record an IQ .wav file so I can give it a try? Thanks.

choupette-change, 2023-08-29:
https://drive.google.com/file/d/1FbggiGkQ1Mis31ueqaxpSR3mY07x6C62/view?usp=drive_link

srcejon, 2023-08-30:
Thanks.

It seems it's only a problem if the "Duplicate channel" button is used. If I use "Add channel"  3 times, then it works.

srcejon, 2023-08-30:
Please ignore the last message - I made a mistake.

If I replay your IQ file with 3 demods, I get:

[image]

What do you get? What do you think is wrong?

choupette-change, 2023-08-30:
I still have the same problem, I don't understand why? I lowered the gain to avoid saturating the receiver (I am near the transmitter)?? Can you send me your settings files to test them with my SDRangel?
[image]

srcejon, 2023-08-30:
Ok, playing around some more, it seems as though there's a bug. Sometimes it works, sometimes it doesn't - even though the settings appear the same in the GUI. Not sure why yet.

choupette-change, 2023-08-30:
Absolutely, I also noticed the random operation of this (these) decoders. Best regards

srcejon, 2023-08-30:
Should be fixed by the above patch.

PR #1795 (2023-08-30): Initialise PhaseDiscriminators state, to avoid outputting huge values.
This PR initialises state in the PhaseDiscriminators class, to prevent it from randomly outputting huge values (ie. 1e+30).

(When such a large value gets in to the MovingAverageUtil, it will stay there, as it keeps a running total, rather than resumming the elements. 1e30-1=1e30)

This fixes #1794.

Patch also connects scope signals in Pager demod, noticed while debugging this issue.