f4exb/sdrangel #1797: BUG: Soapy RSP1A "Global gain" control is reversed.

gvimlag, 2023-08-31:
PROBLEM:
  For the Soapy RSP1A:
    - If "Global gain" INCREASES; then the spectrum display gain DECREASES.
    - If "Global gain" DECREASES; then the spectrum display gain INCREASES.
  The spectrum display gain should INCREASE when the "Global gain" INCREASES, not the reverse.

  To show this problem, I've taken screen shots showing two SDR's:
  - TOP OF SCREEN:    Soapy RSP1A whose "Global gain" is reversed
  - BOTTOM OF SCREEN: Soapy RTL-SDR whose "Global gain" acts as expected

"Global gain" = MAXIMUM (TOP: Soapy RSP1A ---- BOTTOM: Soapy RTL-SDR)
[image]

"Global gain" = 0 (TOP: Soapy RSP1A ---- BOTTOM: Soapy RTL-SDR)
[image]

These screen shots show that the Soapy RSP1A "Global gain" control is reversed,
while the Soapy RTL-SDR "Global gain" control behaves as expected.

  Please correct this "Global gain" control reversal on the Soapy RSP1A.
  Thank you.

SYSTEM:
```
  Hardware-0:   x86 64-bit                        (Client)
  OS-0:         Debian 11.7 (bullseye) 64-bit     (Client)

  Hardware-1:   Raspberry Pi 4B                   (Server)
  OS-1:         Raspberry Pi OS (bullseye) 64-bit (Server)
  SDR-1:        SDRplay RSP1A connected to Raspberry Pi 4B
  RSP1A   Soapy String: soapy=0,driver=remote,remote:driver=sdrplay,remote=<Raspberry Pi 4B Local Server IP address>

  Hardware-2:   Raspberry Pi 3B                   (Server)
  OS-2:         Raspberry Pi OS (bullseye) 64-bit (Server)
  SDR-2:        RTL-SDR       connected to Raspberry Pi 3B
  RTL-SDR Soapy String: soapy=0,driver=remote,remote:driver=rtlsdr,remote=<Raspberry Pi 3B Local Server IP address>

  SDRangel:     (compiled from source following "https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux")
    Version 7.15.3-30-gf891bd681
    Build info: Qt 5.15.2 64 bits
    DSP Rx 16 bits    Tx 16 bits
      Included plugin(s) for "Hardware dependencies":
        SDRplay API
      Included plugin(s) for "Soapy SDR":
        Soapy Remote
```

f4exb, 2023-09-05:
Simple question: why do you use the soapy plugin when RSP1A is supported by the sdrplayv3 plugin natively?

srcejon, 2023-09-05:
Looks like they are remote devices. Perhaps we could make it a little easier to start a remote server without using the API. I will follow up in a separate issue.

As far as this is concerned, I think the SDRplay API gain setting is actually a "gain reduction" setting rather than gain. Not sure how Soapy exposes that, if at all, so could be a Soapy issue.

closed 2023-10-22 by github-actions[bot]