f4exb/sdrangel #1805: ADS-B demod: all airline countries are classified as Taiwan

bczhc, 2023-09-05:
Hi! I just want to politely ask how the country field is fetched. All of them shows Taiwan, but they shouldn't.

[image]

See https://en.wikipedia.org/wiki/List_of_airlines_of_Taiwan; https://en.wikipedia.org/wiki/List_of_airlines_of_China

srcejon, 2023-09-05:
There's a file (demodadsb/flags/regprefixmap.csv) that maps aircraft registration prefixes to the flag.

Looking at it, we have a clash, in that both China and Taiwan both have the prefix B, with nothing to distinguish the two) - and Taiwan is last in the file, so that's why you get the Taiwan flag.

Can you perhaps add another screenshot showing the registration column, so I can see if there's any more info that we could perhaps use?

srcejon, 2023-09-05:
Ok, it seems China should have B-4 digits, and Taiwan B-5 digits, so I may be able to add some additional logic to check for that.

bczhc, 2023-09-05:
Yes. As this: 
[image]
That makes sense. Thanks!

srcejon, 2023-09-05:
Have added a patch - let me know if you still see problems.

bczhc, 2023-09-06:
Alright. Seems to be OK:
[image]

BTW why use the CPC flag but not its national flag? 🤔

srcejon, 2023-09-06:
I just used a flag icon pack from here (https://radarspotting.com/forum/index.php?action=gallery;cat=33). If you have a correct 40x20 pixel image, I can replace it.

bczhc, 2023-09-06:
Oh okay. Actually I have no idea on this, maybe in this case CPC is at the same hierarchy with Hong Kong and Macau (they are Special Administrative Regions)... Just leave it so. Thanks again and this issue can be marked as solved.

closed 2023-09-06 by bczhc